betaflight/betaflight #1961: Support for IRC Tramp channel adjustment through FC

llambkin, 2016-12-31:
Hello all,

I would like to know if it is on the cards to have channel adjustment like TBS Smart audio on the Tramp. Has anyone contacted IRC for their "telemetry" protocol to allow adjustment via the FC/RX

Regards -Liam

raphaelcoeffic, 2017-01-04:
IRC told they would publish/implement it. But it seems that nothing happened since then:
http://fpvlab.com/forums/showthread.php?49370-ImmersionRC-Tramp-HV-5-8TX/page2

llambkin, 2017-01-10:
I just thought Id leave a comment re: work is progressing and people are working on this. It may make it into 3.1

raphaelcoeffic, 2017-01-14:
@llamkim there is now a working version in PR #2106, in case you'd like to try.

llambkin, 2017-01-14:
Thanks Raph,

I'm a bit of noob when it comes to GitHub (I use GitHub Desktop), so I will
wait until it gets accepted. Then I will test.

-Liam

raphaelcoeffic, 2017-01-14:
@llambkin in case you need a binary, I could prepare you one. Just tell me your preferred target. OMNIBUSF4 is already online.

llambkin, 2017-01-14:
Cool I have omnibus f4 in both my quads. One has a spare serial port, the
other needs to use jflypers softserial on pwm5 or 6

closed 2017-07-24 by jflyper