betaflight/betaflight #1962: No roll pid authority REVO betaflight 3.0.1

RBeaubien, 2016-12-31:
Re-Report of https://github.com/betaflight/betaflight/issues/1508 since it was closed.

I have the same issue on 3.0.1. I get 1-3 flights then the motors no longer respond to pitch or roll changes (one or the other). I had this happen to 3 REVO clones and I thought I had bad flight controllers. I have since discovered doing a full erase and reflashing fixes the problem for a while. **I have 2 flight controllers in this state if you want me to test something.** Looking at the sensor data, the gyro appears to be working properly its just the motors do not respond to changes in one direction (pitch or roll).

AndersHoglund, 2016-12-31:
Check you ESC "PPM Min Throttle" setting and compare with FC "min_command". If they are too close or even equal, the ESC might not arm. Make sure FC "min_command" is at least 10 to 20 below the ESC setting.

RBeaubien, 2016-12-31:
Min command is 1000. Min throttle is 1050.  Quad arms correctly.  PID authority over only one of pitch or roll doesn’t work.  All other directions of motion have PID authority.  I have 3 flight controllers experiencing this.  2 of them won’t respond to roll changes, one won’t respond to pitch changes.  Erasing and re-flashing fixes the problem.  This shouldn’t be closed unless it has been found and handled in 3.1.0.  I am a developer and I can help debug this with a little guidance.  I’m just new to STM32Fxx development.

-           Robert Beaubien
-           Sr. Software Architect
-           Kool Software LLC

"No trees were harmed in the sending of this message, however a large number of electrons were terribly inconvenienced."

fftunes, 2017-01-01:
I had very few, and very short, moments where this happened in flight, but it only lasted for like a second or two, losing roll authority only. When flying straight line i wouldn't even notice. 100% sure it was not like failsafe or something since yaw, pitch and throttle reacted perfectly.

RBeaubien, 2017-01-01:
This is not an in-flight issue.  When this occurs, you can't fly as the quad because it no longer tries to stabilize the axis that has failed. This is NOT a receiver/PPM/SBUS issue.  I will post a video showing what happens.  I'm waiting for the weather to clear to test 3.1.0 build 877.

phredieFPV, 2017-01-02:
I ended up giving up on the board that I originally reported this issue about. I thought it must have been a faulty board since no one else seemed to have no one else reporting anything similar. 

@RBeaubien your description sounds exactly like what I was experiencing. Erasing and re-flashing fixed the problem, but only for 1-2 flights.

I still have this board and I am happy to do any testing as well with a bit of guidance.

borisbstyle, 2017-01-04:
can you compare your config before the problem starded and after it occurs.

Also make sure you try the latest firmware (3.1 RC1)

RBeaubien, 2017-01-05:
Sorry guys, I've been sick the past couple of days.  I'm going to get some flying done later on today to see if I can duplicate the problem.  I'll use the latest RC1.

Jahnkeanater, 2017-01-06:
Sounds like the P term is getting set to zero. Maybe there is something in the adjustments tab that is decreasing the P term? Can you check that your pid values are not changing in the configurator?

RBeaubien, 2017-01-06:
No, the PIDs are not changed  They are 43 roll, 58 pitch, and 70 yaw.

RBeaubien, 2017-01-09:
Ok, I flew all day today with RC5 with no issues so I guess the problem was fixed somewhere along the way.

closed 2017-01-09 by RBeaubien