betaflight/betaflight #1969: Why is MAX_MSP_PORT_COUNT only 2?

SandroGrzicic, 2017-01-02:
It seems that for historical reasons (at 2014-08-14, 0b353341f80b81315f584a6fd2def1156eed4b19), the maximum number of MSP ports is 2.

While most of the time this number is enough, some usage scenarios require more than 2 MSPs. For example:

1. USB
2. OSD
3. Wifi module (e.g. https://github.com/me-no-dev/esp8266fcgui)

I have tested increasing this number from 2 to 3 and this has resulted in being able to use the USB port for the Betaflight Configurator, the SP Racing F3 OSD/PDB board, as well as the Wemos D1 mini pro wifi module at the same time.

A developer can easily modify this value for their own local build of BF, but a normal user would have difficulty using 3 MSP ports at the same time. What are the downsides to permanently increasing this value to 3 on a modern F3/F4 board?

Thanks.

andymac4182, 2017-01-02:
I wonder if the gui and wifi chip could end up being built in. With CMS already being there its just another output that would remove the need for a 3rd MSP and create a easy GUI that would be instantly in line with your flashed firmware.

DzikuVx, 2017-01-02:
This is probably some legacy thingie back from BaseFlight. INAV already increased this number to 3 some time ago, and so far everything is working fine.
https://github.com/iNavFlight/inav/commit/3fc177ab6a11afe5342163ce50d96cf7b6b0fdd3

blckmn, 2017-01-02:
Addressed in https://github.com/betaflight/betaflight/pull/1970

closed 2017-01-02 by blckmn