betaflight/betaflight #1973: (low priority issue) Beeper ON_INIT behaves differently on Airbot F4 vs. F3 targets (SPRacing, PikoBLX)

RipperDrone, 2017-01-02:
Minor observation only:

When powering up F4 Airbot FC, the beeper sounds a chirping sequence of init beeps. The F3 types I have (SPR & Piko) don't.

Is this an STM inherent function which cannot be changed or is this part of the BF code. If latter is true, I wonder why on F3 targets there is a beeper ON_INIT switch, but the buzzer still doen't sound any init signal?

MJ666, 2017-01-03:
On some boards the beeper is not powered if you connect via USB only and the beeper is only working with battery connected. Also some of boards can detect the if they are powered by USB via an hardware signal. This likely make the difference in behavior and is part of the individual board hardware.

RipperDrone, 2017-01-03:
Makes perfect sense...

closed 2017-01-04 by borisbstyle