betaflight/betaflight #1975: Infinite spine of death BF 3.1

Jeyfpv, 2017-01-03:
https://youtu.be/T0OuLtq13JQ?t=1m4s
Spins start at 1:08

HW
Alien 5
KISS FC BF 3.1
KISS ESC 24a DSHOT600
F40 v2 2400kv
HQ5040

First time it has ever happened! The quad has been flying extremely smooth till this happened.... now im pretty scared of flying long distances or over the water like i used to.... 
Let me know if i can provide any more info!

borisbstyle, 2017-01-03:
Looks like ESC failed there or one of the propsflew off?

Jeyfpv, 2017-01-03:
None of the above, props were still there after the crash and all 4 ESCs still had their LED light on. Also, i was able to arm the quad when i picked it up. its weird and it has only happened to me once but kinda scary. Def did not sounded like an ESC loosed sync so im not really suru what happened here

borisbstyle, 2017-01-03:
@Jeyfpv 
It may be brown out of your voltage regulator or still the esc's. Gyro connections got loose. It is really hard to say based on just footage. There is not much to investigate based on single reported incident.

I will have to close this issue. In order to have something what could be investigated there are at least few things needed:
1) Blackbox logs of the crash
2) Investigation on the crash site. For example: connect to the board and read all information out of it. verify if it still works and how long is it running and also see if it responds of course.

closed 2017-01-03 by borisbstyle