betaflight/betaflight #1997: Radiolink telemetry support

Jama85, 2017-01-04:
Hello

There is any chance to support radiolink telemetry?

mikeller, 2017-01-04:
From having a look at http://www.radiolink.com.cn/doce/product-detail-110.html, the radiolink telemetry module connects to the FC over a serial connection, which is already something that works with Betaflight.

What it comes down to is the question of what protocol(s) the radiolink module supports, and to either add one that Betaflight supports (MSP / FrSky SmartPort?) to the radiolink module, or add whatever protocol the module supports to Betaflight.

Unfortunately I could not find any detailed specs or manuals for the radiolink module online. Do you have more information about its capabilities?

Jama85, 2017-01-04:
Here you can find some info, but i will ask radiolink.

https://github.com/cleanflight/cleanflight/issues/1690

mikeller, 2017-01-04:
Looks like there is already support for using a PRM-02 module with Betaflight:

> I ended up using Radiolink PRM-02 module with MAVLink telemetry for now. It seeems to me, that the solution, for Naze32 atleast, would be quite non-trivial - it's either wasting 2 ESC ouputs to enable second I2C bus, or implementing I2C in software, or doing some weird multi-master stuff with sensor bus.

Just hook up your PRM-02 to a serial port on the flight controller, and select MAVLINK telemetry for this serial port, and it should work.

Jama85, 2017-01-04:
Yes but would be great to do this without pmr module,cause its too big.
Would be nice if the bf can send data to receiver.

mikeller, 2017-01-04:
From looking at this picture http://www.radiolink.com.cn/doce/admin/editfile/uploadfile/IMAGE/2015-12-02/20151202141359150.jpg , the PMR-02 module seems to be a fraction of the size of the smallest RX that it can be used with, so eliminating the PMR-02 will not solve your size problem.

If size is a problem, maybe you should look into using a FrSky micro RX with telemetry?

But anyhow, connecting radiolink RX telemetry to the flight controller through I2C is another option. So far, the protocol used by the PMR-02 has not been published or reverse engineered. Once one of these things have happened, adding support for it to Betaflight will be possible.

jflyper, 2017-01-05:
Just my 2 cents:
Directly connecting Radiolink telemetry over I2C would be non-trivial, because it requires I2C slave code on the FC side.

closed 2017-01-22 by Jama85