betaflight/betaflight #1998: autotune feature

Jama85, 2017-01-04:
Hello

There is any chance to make autotune function, by fc? maybe use blackbox log.

robert-hoffmann, 2017-01-10:
Would be cool !
Librepilot backported dronin's autotune, you can check it ou here
https://bitbucket.org/librepilot/librepilot/src/cf1b64966e4d4ca13d6f55f3c17815fcb8cb6242/flight/modules/AutoTune/?at=next

airuno2L, 2017-01-22:
I think manual PID tuning is one of the biggest barriers for entry for newcomers and also think this would be a great feature.  

I'm wondering if anyone has experience with or knowledge of the history of past attempts at autotuning for quadrotors. Ive read a little about them and know there have been many attempts - utune, gtune, optune, autotune but I've not used any. Is one them the "best"? If not, where did they fall short?

From my limited reading it seems like they're all heuristic tuners (maybe autotune is the exception?). I would think a method that uses system identification would be the best bet. Like comparing controller input to gyro measurements and making a first order plus dead time (FOPDT) model which can be used to generate good PIDs. 

Its believe it would be possible to monitor a regular flight (closed loop) and automatically figure out the parameters for the FOPDT model (i.e. not require open loop step responses).

My day job is ground robotics and I have a bit of background in this. I haven't dove into clean or betaflight code before and my time is limited, but I could help with implementation. Does anyone know of anything that would prevent that method from working well?

robert-hoffmann, 2017-01-24:
Well personally, the only reason i've not transitioned to betaflight yet is because the autotune in Dronin DOES work pretty nicely (at least at my level piloting).

However i've heard that even without any tuning Betaflight manages extremely well on default values (at least on F3/F4 targets), so maybe i'll try the jump anyways... Especially now that 3.1 is out :-)

That said, if the BF/CF team could get autotune in, that would probably be the ultimate killer function to get everyone onboard, regardless of their level of expertise. ..well maybe not everyone, dronin/librepilot does have a nice setup wizard that caters rather niceley to noobs

sudo-robot-destroy, 2017-01-27:
I've got a few ideas I'd like to start working on but I don't have a flight controller with blackbox at the moment. Is there a place where I can download some blackbox logs? 

Ideally they would be a recording of someone hovering stable, then hitting a hard roll one way then a hard roll the other way, then going back to stable hover. But I'll take anything I can get my hands on to get started. Something else that would be useful is PID values that work well for the vehicle being recorded.

sudo-robot-destroy, 2017-01-27:
Right now my plan is to do an offline FOPDT fit because it seems like the easiest thing for me to start with. It could change in the future but I'd rather make sure the idea works before getting into the implementation details. 

So the procedure would go like this:
1. In acro mode, begin recording blackbox
2. The pilot takes off and goes to a hover (ideally there would be no wind and no drift - the trim would be set good)
3. Then the user rolls hard to the right, then back hard to the left to get back level and stabilize the quad.
4. Repeat for pitch and yaw...I don't know if it will be easy to do all axes in one flight and pick out the separate maneuvers from the log or if it will be easier to just do separate flights and separate recordings for each axis.
5. Take the logs to the blackbox viewer. I'm guessing the fitting software should reside in the viewer since it operates on the logs.
6. Do a non linear least squares fit do get the parameters to a FOPDT models that maps the recorded controller output data to the recorded gyro data. 
7. Once the FOPDT is known, the PID values can be generated similar to how it's done here: http://controlguru.com/pid-control-of-the-heat-exchanger/. Each axis would be its own FOPDT.

sudo-robot-destroy, 2017-02-03:
I ran into an unanticipated problem. I turns out that rate control of a quadcopter is what is call a non-self-regulating (aka integrating) process. 

There is a good explanation of what that means [here](http://controlguru.com/recognizing-integrating-non-self-regulating-process-behavior/), but in short, a normal self-regulating process is one where if the controller output (PID_sum) and disturbances (wind) are held constant at fixed values, the process variable (gyro measurement) will be steady at a constant value. If we increase the controller output, wait a bit, and then return it to its original value, the process variable will respond during the experiment and then return to its original steady state. But a non-self-regulating process doesn't do that. 

I think of a self-regulating process as having a proportional relationship between the controller output and process variable, but a non-self-regulating process as having an integral relationship. Here is a picture showing the difference: 

[image]

I've never dealt with non-self-regulating processes and they're harder to model and control. But modeling it comes down to trying to fit recorded data to the following discrete model (I think it is correct but I'm not sure yet, I'm rusty on my calculus): 

`PV(n+1) = PV(n) + (1 - e^-T)*Kp*CO(n-theta_p)`

Where:
PV(n+1) is the process variable (i.e. gyro measurement) value for the next time step 
PV(n) is the process variable value for the current time step
T is the sampling rate (i.e. time between samples)
Kp is the integrator gain
theta_p is the dead time, or the amount of time it takes for the system to respond to a control action
CO(n-theta_p) is the controller output (i.e. PID_sum) that was issued a little while ago. 

The optimization problem is fitting the above equation to the recorded data by trying different values for theta_p and Kp. 

I have python code that I think works, but I don't have any good data to test it on. I tried grabbing some random logs from places online of people flying around and testing it, but there are too many disturbances present in them. I really need a log where someone is sitting still hovering then rolls hard to the right (for example up to 45 degrees), then back hard to the left to get back level and stabilize the quad. 

Logs where someone is flying around aren't working because all the wind and air interactions cause fluctuations in the controller output to reject them and that effects the mapping of controller output to process variable. 

I'm going to ask around for someone to generate a suitable log file, I don't think I'll have a blackbox capable flight controller anytime soon.

robert-hoffmann, 2017-02-04:
Why don't you just generate the roll/hover/etc yourself programatically ? Thats how autotune in dronin works, it generates (known) movement, and then calculates from what (actually) happened.

sudo-robot-destroy, 2017-02-04:
The honest reason is I'm a much better python programmer than C haha. Right now everything I'm doing is standalone, it just takes in a cvs log file as input. I don't know anything about betaflight code yet.

Generating movements programmatically is definitely something to consider in the future, but I just want to make sure the idea works first. I do think as a user I'd rather generate the movements myself rather than them be done automatically, but that might just be me. I like to have control.

I'm not sure, but I assume the reason autotune did it programmatically is because they needed step input responses, but with the least squares method you don't need that, you just need recorded gyro and controller output data during movements with magnitudes well above noise levels and without too much disturbances involved.

ledvinap, 2017-02-04:
@sudo-robot-destroy : I am afraid you are greatly underestimating the problem. 
- Quadcopter is highly nonlinear system
- There are large external disturbances and lot of noise
- You are working closed-loop process. Getting useful data from it may be non-trivial (you usually need open-loop/plant model to optimize controler)

Maybe you should start with some textbook model (inverted pendulum is good candidate), simulate it and try your algorithms on simulated data.

I played with relay autotuning some time ago, using relay on angle and angle rate. That way it is possible to keep quad flying (using angle relay is not enough ... was quite fun finding out) and gather quite open-loop system data (input is constant for some time).
I did try system identification toolbox from Octave to estimate system parameters, it did produce some results, but nothing spectacular.
I'll look for my data ...

ledvinap, 2017-02-04:
Here is one of may relay logs. Only documentation is my source code (local branch is in non-working state now): https://github.com/ledvinap/cleanflight/blob/local/src/main/flight/autotune_relay.c
This one should match latest relay code. 
[LOG00128.zip](https://github.com/betaflight/betaflight/files/752455/LOG00128.zip)

sudo-robot-destroy, 2017-02-05:
Hi ledvinap, thanks I'll take a look at that. 

I've used the method I'm describing on real world equipment before, it was a speed controller for a brushed DC motor with an optical encoder on a welding robot. Although that system was self-regulating and it was easier to ensure disturbances were minimized during recording of the tuning maneuvers, so it was a little easier.

I've got some results for my model fitting attempts that look promising, I'll try to post them soon.

Dronek, 2017-02-09:
Guys, autotune was actually present in betaflight and was intentionally removed.
Just in case, you missed it:
[](https://github.com/betaflight/betaflight/issues/339)

I gave it a try in [cleanflight](http://shipow.github.io/cleanflight-web/docs/autotune.html) but in my opinion you're better off with BF 3.1 on stock settings than any auto-tuned result back then. It would be more reasonable to think about alternatives to PID control or let neural networks tune our quads some day, maybe ;)

borisbstyle, 2017-02-09:
@sudo-robot-destroy @ledvinap 
Using "users input" is never going to be exact science.
What I used to do is write a code that does a sequence of commands and measuements which all get logged into blackbox as events. I used it by attaching a multirotor on strings or hovering before I hit the test switch.
It looks something like this. On this way you can test the true response times on the mini multirorotors and do some real modelling
[image]

But as @ledvinap already says the multirotors are suffering from so many variables that perfect automatic tuning is almost impossible.
In fact most tuning nowadays is all about the cosmetic tuning rather than getting the best response from system and there is a difference between those.

closed 2018-04-21 by DieHertz