betaflight/betaflight #2000: feature request: link quality alternative rssi  (osd and blackbox)

DeepnineRC, 2017-01-04:
Hi,

i would like to see link quality on sbus receivers based on frameloss per second.

sbus frame = 9ms
while a frame is lost, the values of the last valid frame are sent.
100 frames lost in a row triggers a failsafe on frsky receivers (bus_failsave_flag).

1.) frameloss per 0.9 second (100 frames) 
or
2.) frameloss in row per 0.9 second (100 frames)

mikeller, 2017-01-04:
@DeepnineRC Is there any indication sent by the RX when a frame loss happens before the `bus_failsave_flag` is triggered (except for the repeat of the identical frame, which could also be caused by no input change on the TX)?

jflyper, 2017-01-05:
IIRC, there is a receiver-dependent slot at the end of each frame indicating "some" sort of rssi or plr.

schugabe, 2018-09-27:
https://github.com/betaflight/betaflight/pull/5090 added a rssi_src_frame_errors option and with this it uses the detection of dropped frames from the sbus protocol and displays it as rssi. 

If https://github.com/betaflight/betaflight/pull/6480 gets merged this number is available as link quality and not rssi.

I'd say that the two PRs provide a solution for this feature request.

closed 2022-05-09 by haslinghuis