betaflight/betaflight #2003: Buzzer function not working on REVO F4 Clone

Twinkle9, 2017-01-05:
After upgrade BF 3.1 RC1, Buzzer is always beeping on acro mode and no sound on angle/horizon mode..
also not working arming / warning / manual beeper sound.. I used betaflight_3.1.0_REVO.hex
It was working fine on BF 3.0.1 previous..
[image]
[image]

blckmn, 2017-01-05:
That looks like the AIRBOT (a derivative of the REVO, but with different buzzer setup)

Can you please try again using that target.

atomiclama, 2017-01-05:
Yep I use the AIRBOT build on that HW, buzzer works fine. 

You must also do a default to get the correct resource mapping once you have switched to the AIRBOT build, otherwise it will use the revo mapping which is incorrect for that HW, unless this has been changed in a recent build.

closed 2017-01-05 by blckmn