betaflight/betaflight #2007: (pikoblk) led_strip and dshot

frizzle101101, 2017-01-05:
currently the led_strip and motor 4 share the same dma channel 3. causes motor 4 not to work when led_strip is enabled.

blckmn, 2017-01-05:
This will be a limitation of the FC - if there is not an alternate timer / channel available for the motor 4 pin, then you will not be able to use the LED_STRIP and DSHOT at the same time unless:

You remap led strip to an alternate output, or remap motor 4 to an alternate output.

jflyper, 2017-01-05:
Wasn't there a mod to use ppm pad for motor 4?

blckmn, 2017-01-05:
The mod was to remap motor 1, so that dshot was suitable - but this is for getting ledstrip AND dshot running.

On this target it may not be possible at all.

jflyper, 2017-01-05:
@blckmn It is impossible to run LED strip AND Dshot running at the same time.

[image]

Micro soldering, may be.

blckmn, 2017-01-05:
I agree this is a hardware limitation :(

closed 2017-01-05 by blckmn

frizzle101101, 2017-01-05:
Y not motor signal 5 it uses channel 7?

jflyper, 2017-01-06:
If BLX has PA1 broken out as solderable-pad, yes.

frizzle101101, 2017-01-15:
SO if one looked into this before saying it was impossible as i did then they would have dshot and leds working on their micro quad. To give info for this topic I will outline the steps and reallocation needed. To start, to allow dshot on the pikoblx target motor 1 has to be moved to ppm pin a07 to use dma1 ch1. Also if one wants capability of using the led_strip pin, motor 4 must be moved to motor 5 pin a01 to use dma1 ch7 while the led_strip uses ch3 which motor 4 uses. The resource pins can be changed in the betaflight cli.

mikeller, 2017-01-15:
@frizzle101101: The first half of this (without LED_STRIP) is already documented in https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol . Could you please add your findings on how to make LED_STRIP work to the entry for PIKOBLX?