betaflight/betaflight #2009: configurator disconnects when using aux channels

dimant, 2017-01-05:
100% repro, worked with betaflight 3.0.1 + configurator 1.8.5, doesn't work with betaflight 3.1 rc + configurator 1.8.5

Taranis plus, X4R black receiver, spracing f3 evo

1. usb vcp -> blackbox + msp; uart1 -> msp (for osd); uart2 -> serial rx; uart3 -> smartport telemetry
2. set TAER mapping in receiver tab
3. go to modes and add arm mode
4. connect usb + battery
5. flip receiver switch

expected: arm mode is highlighted yellow, configurator stays connected
actual: configurator loses com connection

on reconnect everything works and the mapping is saved correctly to the FC (I can arm my quad)

mikeller, 2017-01-05:
This sounds like it is by design. If you assign MSP & Blackbox to the VCP, the port will switch from MSP to 
Blackbox when arming. See https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md :

 > MSP/CLI can be shared with EITHER Blackbox OR telemetry. In shared mode blackbox or telemetry will be output only when armed.

dimant, 2017-01-05:
This sounds very convincing and I would accept this explanation if the behavior was the same in 3.0.1 - the previous behavior was that blackbox would only log when USB <-> PC was disconnected. Perhaps someone familiar with the design decision can chime in on whether this is a bug or a bugfix?

dimant, 2017-01-05:
just reflashed to 3.0.1 and you're right - the behavior is the same. Must be seeing ghosts - my apologies.

closed 2017-01-05 by dimant