betaflight/betaflight #2016: No MSP over SPORT with OmnibusF4

DerDominik, 2017-01-05:
Hi there

I wanted to test the MSP over SPORT function with my OmnibusF4 FC to configure PID settings with my Taranis. But no values will be displayed. Telemetry seems to be ok, cause I get voltage, current, ... values, and inside of the script there is no "No telemetry" message.
Taranis is on v2.2N360

With almost same config the function works with my SPRacingF3 FC.

Anyone tested this function with OmnibusF4 FC?

raphaelcoeffic, 2017-01-05:
That looks damn strange to me. It should work in fact. Unfortunately, I do not have any working F4 board to test it (whereby I'd love to ;-)).

SoulPictures, 2017-01-05:
@raphaelcoeffic maybe i can help out in the next days ;) mine is on the way and the setup will include an BlueJayF4 with XSR, UnifyRace etc. but Horus ;)

raphaelcoeffic, 2017-01-05:
> Is there something, what I can test to find out the failure?

Yes, you could test the other script, and post the numbers that will appear on the screen.

DerDominik, 2017-01-05:
[image]

Attached is a picture after short using of the testscript.
I think, this looks good?!

raphaelcoeffic, 2017-01-05:
Yes indeed. I can try to re-test with your OTX version. Let's see what happens. Anyone else having troubles with the LUA script?

DerDominik, 2017-01-05:
I think the OTX Version is not the problem due it is working with my SPRacingF3

raphaelcoeffic, 2017-01-05:
@SoulPictures should be able to help this week end. We'all probably meet for some testing with his F4 board. Let's see what comes out of it.

raphaelcoeffic, 2017-01-05:
@DerDominik in case you own a logic analyser (saleae or clone), could you possibly send a trace of happens on the Smart Port?

DerDominik, 2017-01-05:
I'm sorry, but directly I don't have a logic analyzer. And a scilloscope doesn't help so much?!
But maybe a Arduino is possible for tracing?!

Eganwp, 2017-01-06:
If I have enough time tonight I'll test this out on some F3 and F4 boards. A big if on the F4 as I still have a few on the bench unwired! :(

raphaelcoeffic, 2017-01-06:
We have a couple of people who already tested on F4, and more specifically on the Omnibus F4 Pro (using opentx 2.2.0N360 via xsr with non-inverted smartport hack).

This issue seems hard to reproduce somehow...

raphaelcoeffic, 2017-01-06:
We could see it working ok with a BlueJayF4 (rev3) as well.

SoulPictures, 2017-01-06:
[image]

Works fine on BlueJayF4 with RC4 // Taranis 2.2.0 N360 // XSR with non inverted Hack

DerDominik, 2017-01-07:
Thats strange. In my case I use a X4r with using an internal S.Port Pin due the invertation. But I don't think that should be the reason as the 'normal' telemetry is working

raphaelcoeffic, 2017-01-07:
Do you mean that you tapped the non-inverted signal on the X4R? I did the same in fact to connect it to a Naze hardware serial and developed the bridge on that hardware. So I think it should work the same. Could still post some pictures? Just in case...

DerDominik, 2017-01-08:
[image]

Attached you'll see a picture from the X4R
But back to config of this feature. Is there something special to configure it? Or should it automatically work, if telemetry is working?
Maybe it is important. I'm using UART3 on the OmniBusF4

raphaelcoeffic, 2017-01-08:
@DerDominik it should work as soon as SPORT telemetry is working. No specific serial port is required.

raphaelcoeffic, 2017-01-14:
@DerDominik do you have anything new? Did you finally get it to work?

DerDominik, 2017-01-14:
Tested it again with RC7.
I've loaded default setting, and then configure only SBUS und Telemetry over S.Port
But the script doesn't work :-/ But if that script works on other F4 FC, then I think there is a problem with the Omnibus?!

raphaelcoeffic, 2017-01-14:
@DerDominik what stomachs me is that is has been tested successfully with the OMNIBUSF4 as well. Have you tried to use another UART for SPORT?

SoulPictures, 2017-01-14:
Wierd thing which happened to me here with my F4 was that it was not working after flashing RC7, S.Port Telemetry wasn't working neither. Went to CLI, save my min_check stuff, saved and just Rebooted the whole thing through bat power instead of USB and it started working. Don't  know what happened there, since then its working here...

raphaelcoeffic, 2017-01-14:
@SoulPictures did you use "Full chip erase" when flashing RC7?

SoulPictures, 2017-01-14:
Yup Full chip erase was used

raphaelcoeffic, 2017-01-14:
@SoulPictures that's pretty weird!

SoulPictures, 2017-01-14:
kinda like smart port is only started correctly if RX has power at FC boot

raphaelcoeffic, 2017-01-14:
@SoulPictures right, if smart port cannot be initialised after boot (happens when powered with USB, and the RX is not powered), then the FC gives up on smart port for the rest of the time (no retry). This could be changed though. Frankly, I don't know why it has been implement this way.

DerDominik, 2017-01-14:
@raphaelcoeffic
It's hard to test it on another UART, as it is hard connected ;-)
But I've another OmnibusF4. I'll try to build a dirty system with possible to quick change
General question: Does the script work with connected USB on F4? On my SPRacingF3 it is only working with disconnected USB

SoulPictures, 2017-01-14:
my F3 test was always via USB, my F4 via Bat

raphaelcoeffic, 2017-01-14:
> 
> @raphaelcoeffic
> It's hard to test it on another UART, as it is hard connected ;-)
> But I've another OmnibusF4. I'll try to build a dirty system with possible to quick change
> General question: Does the script work with connected USB on F4? On my SPRacingF3 it is only working with disconnected USB

It's better when SPORT does not share the UART with MSP. SPORT in General does not work very well when the receiver is not powered at the same time as the FC. In this particular case, SPORT is disabled until new power cycle.

DerDominik, 2017-01-14:
I tested it with my other OmnibusF4, but the same error :-/
Now I connected only smartport with power supply to UART1 (same input on the X4R as on the picture above). Telemetry is ok, but on the Script I got no values.

What I'm wondering, if I connect UART1/TX directly to the Smartport from the X4R I got no telemetry. But, as far as I know, UART1 is possible to do the telemetry inversion?!

SoulPictures, 2017-01-14:
As far as i can see there is only one inverter for RX on UART1 non for other UART oder the TX Line.... Information from here: http://www.readytoflyquads.com/flip-32-f4-omnibus

DerDominik, 2017-01-14:
OK, then it's ok for that problem ;-)
But....
On UART1 the script also doesn't work

DerDominik, 2017-01-14:
On your pull request for the Horus-Script you wrote:

> [WARNING] At the moment, the latest nightly OpenTX available (N360) still has an issue with sending SPORT packets from LUA scripts (opentx/opentx#4205). However, this is already fixed in "next" branch however.

Maybe this is here also the same problem?

raphaelcoeffic, 2017-01-14:
No, taranis has always worked good. OTX 2.2 N360 is good enough.

DerDominik, 2017-01-15:
Hm...
What possibilites do I've to test this a little bit deeper? It is possible to record the data traffic from the Taranis to the SD Card for further analysis? Or only with a logic analyzer?

@raphaelcoeffic 
You wrote, that it has been tested with the OmnibusF4? Who tested it? And which RX was used?

SoulPictures, 2017-01-15:
@DerDominik if you are in Berlin we could do a little local debugging

DerDominik, 2017-01-15:
Not really ;-) I'm living in Hameln, near to Hannover

raphaelcoeffic, 2017-01-15:
> 
> Not really ;-) I'm living in Hameln, near to Hannover
Just contact me by email, we can probably arrange something (rco@iptel.org).

DerDominik, 2017-01-24:
Issue can be closed - Problem was, that a newer lua-script was not taken inside of the taranis modell.
Don't know why, but maybe a version number display inside of the script would be great :-)

closed 2017-01-24 by DerDominik

Saxin, 2017-07-07:
Can you please advise what was the issue I'm having exactly same behavoiur, one model on the taranis @ 2.2.0 , F3 works , x2 Omnibus F4 - NO GO.