betaflight/betaflight #2017: feature request: OSD icons for flightmodes

skaman82, 2017-01-05:
I just discussed with @jflyper
Can we please add some icons for flight modes instead using simply a text? 
Just like in MWOSD. 

Icons for: 
• STAB
• HORZN
• ACRO
• AIRMODE

jflyper, 2017-01-05:
STAB is easy... A balance... 
[image]

HORIZON
[image]

ACRO
[image]

AIRMODE
[image]

DanNixon, 2017-01-06:
Is there a problem with just text? Especially when all MWOSD has is text in a box.

The advantage is that it does not take up extra character space in the overlay IC and the text is larger making it easier to read (good for folk like me that usually wear glasses).

skaman82, 2017-01-13:
@DanNixon I find it way to large and find the font icon (like it is displayed in MWOSD) much nicer. I also wear glasses :)

jflyper, 2017-01-14:
Once we have font glyphs, we can make it switchable between letters and icons.