betaflight/betaflight #2018: MSP_VTX_CONFIG impl. or VTX control over MSP over S.PORT

jflyper, 2017-01-05:
I remember someone using a lua script to control SmartAudio.
Was the functionality ever submitted?

Refs:
MSP/SPORT: BFSetup LUA script fixes #1608
Frsky Smart Port / MSP bridge #1401

@raphaelcoeffic?

raphaelcoeffic, 2017-01-05:
It was not, but I can ask the guy. I think he was waiting for the SmartAudio code to be merged.

raphaelcoeffic, 2017-01-05:
Yes, but I guess the MSP_VTX_CONFIG should be around in his repos as well.

jflyper, 2017-01-05:
@raphaelcoeffic Ack. I think it would be better to bang MSP_VTX_CONFIG interface.

@skaman82 That's it. How did it get in there? And I also believe there must be matching code on the fc_msp.c side...

raphaelcoeffic, 2017-01-05:
The script can probably use the new interface made for the configurator, right? I guess there is such, right?

jflyper, 2017-01-05:
@raphaelcoeffic No, the configurator only configures an UART for the SA.

raphaelcoeffic, 2017-01-05:
@jflyper ok, no problem. I'll be pleased to integrate it into LUA when the MSP part is done. It seems to be need for the configurator anyway, according to the ongoing discussions.

raphaelcoeffic, 2017-01-23:
#2146

closed 2017-01-26 by jflyper