betaflight/betaflight #2019: 3.1 Omnibus OSD menu not showing fully (Configurator 1.8.5)

deverlight, 2017-01-05:
I'm having the issue with my Omnibus F3 on 3.1 where the OSD screen in the configurator is only showing a very thin bar across the top.  I've seen the previous times this issue has been posted where updating to the latest configurator seems to fix the issue.  However, even 1.8.5 does not fix the issue for me.

[image]

jflyper, 2017-01-05:
You need an up to date configurator.

You can force use it by:

(a) Goto Betaflight Configurator github.
(b) Press the green "Clone or Download", select "Download ZIP", save and expand it at your favorite location.
(c) In Chrome, open "Extensions Manager", check "Developer Mode" and press "Load unpacked extension...".
(d) Select the location you have saved your files in (b).
(e) Launch.

deverlight, 2017-01-05:
Thought 1.8.5 was the most recent configurator.  Looks like there was an update within the last few days.  Seems to work fine on 1.8.6 - thanks!

closed 2017-01-05 by deverlight