betaflight/betaflight #2020: Adding Softserial to AIRBOTF4 target

cephyr, 2017-01-05:
INAV supports SOFTSERIAL on the AIRBOTF4 target (subtarget of REVO)

Is it possible to get this feature on betaflight too?

they use the following pads on the board:
[https://d2obrsbtn4fa9r.cloudfront.net/wp-content/uploads/2016/12/Airbot-F4-SoftwareSerial.jpg](url)

I've tried to add the inav code to the betaflight target but the syntax in their target.c files seems to be different

jflyper, 2017-01-05:
A PR exists for this, not only for AIRBOTF4 but for all targets, as a part of re-configurable (re-mappable) resource management:

Configurable serial port pin assignment #1849

It's not final, but I hope will be elaborated and be included as a part of the phase 2 resource management in a future release.

In a mean while, there is already a hex for AIRBOTF4 at [Exotic OMNIBUS](https://www.rcgroups.com/forums/showthread.php?2792346-Exotic-OMNIBUS), with SOFTSERIAL1 configured on S5_IN and S6_IN (I believe they are identical to the iNav assignment, because this is what I proposed). The hex is bit old, so I will try to add another set that is based on RC1 + #1849 (and possibly with #1867 for maximum configuration flexibility).

nathantsoi, 2017-04-25:
did you get it working @ZeeZeeEff? mind closing this issue if you're good to go?

ZeeZeeEff, 2017-04-25:
Hey,
Yes the issue is solved, thank you! 

---- Nathan schrieb ----

closed 2017-07-24 by jflyper