betaflight/betaflight #2028: gyro.c - warning/note output when compiling

TheAngularity, 2017-01-06:
@martinbudden 

pls can you see on this ... imho it comes with latest #2023 
```
%% stm32f4xx_gpio.c
./src/main/sensors/gyro.c: In function 'gyroUpdateISR':
./src/main/sensors/gyro.c:390:18: warning: passing argument 1 of 'alignSensors' discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
     alignSensors(gyroADC, gyroDev->gyroAlign);
                  ^
In file included from ./src/main/sensors/gyro.c:60:0:
./src/main/sensors/boardalignment.h:26:6: note: expected 'int32_t * {aka long int *}' but argument is of type 'volatile int32_t * {aka volatile long int *}'
 void alignSensors(int32_t *dest, uint8_t rotation);
      ^
./src/main/sensors/gyro.c: In function 'gyroUpdate':
./src/main/sensors/gyro.c:437:18: warning: passing argument 1 of 'alignSensors' discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
     alignSensors(gyroADC, gyro.dev.gyroAlign);
                  ^
In file included from ./src/main/sensors/gyro.c:60:0:
./src/main/sensors/boardalignment.h:26:6: note: expected 'int32_t * {aka long int *}' but argument is of type 'volatile int32_t * {aka volatile long int *}'
 void alignSensors(int32_t *dest, uint8_t rotation);
      ^
%% stm32f4xx_spi.c

```

ledvinap, 2017-01-06:
'volatile' is bad idea .. 
in this case ATOMIC_BLOCK (or better ATOMIC_BLOCK_NB) in non-isr code will handle memory barriers properly and provide proper protection .. And ATOMIC_BARRIER can be used in ISR handler, but in most cases no protection is necessary

ledvinap, 2017-01-06:
for example: http://stackoverflow.com/questions/2484980/why-is-volatile-not-considered-useful-in-multithreaded-c-or-c-programming

borisbstyle, 2017-01-06:
@ledvinap Do you suggest to remove it entirely now?

ledvinap, 2017-01-06:
It's not that simple ... volatile does mask most of problems with variables assessed from ISR (in some cases by reducing probability of problem and thus creating hard to reproduce bugs)

All access to data shared between ISR and main code shall be protected somehow. 

ATOMIC_BLOCK is one way to do this, it creates memory barrier, so all variables must be read from memory  and changes must be written back before ATOMIC_BLOCK ends. ATOMIC_BLOCK_NB + ATOMIC_BARRIER does the same, but memory barrier is explicit, so unprotected variables are kept in registers (gcc 5+, generated code is considerably better).

ISR handler function has implicit memory barriers on entry and exit (it is called from 'nowhere', it can't be inlined etc). Also interrupt priority will usually prevent concurrent access. Because of this, no special care is usually necessary.

Queue handling is partial exception when ISR handles always accesses one side and main code the other. All that is necessary is that integer read/write is atomic (which is true on ARM, but not necessarily on other processors)

borisbstyle, 2017-01-06:
@ledvinap thanks for the link and explaination.
In current approach we have experimental and disabled feature where gyro read occurs within the ISR. As this is disabled by default anyway just a volatile is fine for now.

closed 2017-01-06 by borisbstyle