betaflight/betaflight #2031: no VFAS reported over FrSky Telemetry for 1S

Adminiuga, 2017-01-06:
Hi All,
I have a QX90 which was updated to use FrSky Telemetry (mostly for VFAS). I do get VFAS on my taranis on Betaflight 3.0.1, but it does not work on Betaflight 3.1.0-RC4. Per git bisect the ceeef947d5c42e474692be872cdda4e24edb975e is the 1st bad commit.

I do get other sensors like AccX/Y/Z, vSpd, Hdg etc and I do see the updates for those sensors, so telemetry is working. I've tried to delete all sensors and rediscover them again, which brings all sensors back but the VFas. BF configurator also is showing correct voltage. and if I flash v3.0.1 back, then I do get VFas in my telemetry.

Quad: QX90
Version: 3.1.0-RC4
config dump: `# version
# BetaFlight/SPRACINGF3EVO 3.0.1 Oct 18 2016 / 10:36:44 (48b7b4f)

# name
name QX90

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature -CURRENT_METER
feature -RSSI_ADC
feature -BLACKBOX
feature -TRANSPONDER
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE

# beeper

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 4 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1750 2100
aux 1 1 3 1725 2100
aux 2 2 3 1300 1625
aux 3 26 0 900 1125

# adjrange

# rxrange

# rxfail
rxfail 6 s 1500

# master
set debug_mode = NOTCH
set min_throttle = 1020
set motor_pwm_protocol = BRUSHED
set serialrx_provider = SBUS
set deadband = 5
set yaw_deadband = 5
set failsafe_kill_switch = ON
set acc_trim_pitch = 10
set pid_process_denom = 8

# profile
profile 0

set dterm_setpoint_weight = 104
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 70
set roll_srate = 68
`

Adminiuga, 2017-01-06:
**Update**
Apparently the VFAS and Cels do get reported when the copter is armed, but as soon as I disarm, those are not being reported any longer and I get "Sensors Lost" audio alarm on my Taranis (OpenTX 2.2). I need to think if this is the right behavior or not.

wind0r, 2017-01-06:
You use a 1S lipo or? With the default settings battery monitoring is only activ when armed or when a battery over 5.5V is pluged in.

If you want to change that behavior just use the CLI and write "set battery_notpresent_level = 3.3"

Important Lines:
https://github.com/betaflight/betaflight/blob/master/src/main/sensors/battery.c#L150
https://github.com/betaflight/betaflight/blob/master/src/main/fc/config.c#L426
https://github.com/betaflight/betaflight/blob/master/src/main/fc/serial_cli.c#L590

Adminiuga, 2017-01-06:
hm, so I've tried with "set battery_notpresent_level = 33" and "set battery_notpresent_level=56" but I still do not get VFAS sent over telem when disarmed and therefore getting "sensor lost" audio alarm on taranis :(  still trying it on v3.1.0-RC4

Adminiuga, 2017-01-06:
Nevermind, I've must have screwed something, like not connecting the battery and running of the USB only. 

So with "set battery_notpresent_level = 33" and battery connected I do get VFAS sent when disarmed. And if I unplug the battery, then vfas is not being sent any longer and I get "sensor lost" audio alarm.

I think this is an acceptable behavior, although I see how it may spook some users on battery disconnect. 

So was it done for the purpose of not getting an "critical voltage alarm" on battery disconnect?

DanNixon, 2017-01-06:
> So was it done for the purpose of not getting an "critical voltage alarm" on battery disconnect?

Usually you'd just get "telemetry lost" at this point (unless the FC/RX is powered over elsewhere).
Is the "sensor lost" alarm an OpenTX 2.2 thing? Don't think I've ever had that before (and I have had sensors stop working).

Adminiuga, 2017-01-06:
> Usually you'd just get "telemetry lost" at this point (unless the FC/RX is powered over elsewhere)

You are right. I was overthinking it. On the bench FC & RX is powered by USB and LiPo, in the field I'd just get telemetry lost. 

> Is the "sensor lost" alarm an OpenTX 2.2 thing? Don't think I've ever had that before (and I have had sensors stop working).

Looks like it could be OpenTX 2.2 thing. Never heard it before, but on the other hand I'm not sure if I had only some specific sensors stop working.

Closing this issue. Thanks to everyone for the great work and I'm very happy to have voltage monitoring now on my micro drone :)

closed 2017-01-06 by Adminiuga

famtory, 2017-07-20:
I have same problem.

slash4x4, 2017-11-15:
I also have the same problem anybody know how to fix it running the Lux V2 flight controller s bus is on uart 4 i have S. Port hooked up to uart 5. It was working fine on betaflight 3.1 6 flash to the newest 3.2 2 and it no longer get telemetry anybody know what's wrong?????