betaflight/betaflight #2032: Led strip and beeper issue on REVO board

GrishaLutsenko, 2017-01-06:
So, right now flashed and got problem that beeper is always ON (soldered to BUZ+ and BUZ- pads). Before flashing 3.1.0 RC4 all was fine

Also LED strip is not on LED_STRIP pad, for some reason it is on motor 5 output...

GrishaLutsenko, 2017-01-06:
Board is F4 REVO clone. Simple black one 
http://www.banggood.com/Raceflight-Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-with-VbatBuzzer-p-1087561.html?rmmds=search

GrishaLutsenko, 2017-01-06:
Of, flashed AIRBOTF4, buzzer fixed, but LED strip is still at motor 5 out

jflyper, 2017-01-07:
@GrishaLutsenko Yes. On REVO and AIRBOTF4 targets, LED strip is on motor 5 out.

blckmn, 2017-01-10:
Closing issue as this is by design. Resource command can remap led strip to alternative pin if desired.

closed 2017-01-10 by blckmn

free6k, 2017-06-08:
@blckmn, hello, how remap motore 5 to ledstrip? thanks

mikeller, 2017-06-08:
@free6k: Read https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping .

free6k, 2017-06-09:
dont work this solution..
resource LED_STRIP 1 A01
resource motor 5 NONE
save

LEDSTRIP dont work... Or Ledstrip have another name for this board? dont work this solution..
resource LED_STRIP 1 A01
resource motor 5 NONE
save

LEDSTRIP dont work... Or Ledstrip have another name for this board? Who has a solution? Thanks for all!

jflyper, 2017-06-10:
@free6k Association of solderable pads and pins are different from board to board. You must know your board's association.