betaflight/betaflight #2036: 3.1.0 rc1, rc4 CC3D_OPBL cannot startup.

Sheynar, 2017-01-07:
Walker Runner 250c (cc3d) main controller. Flashing 3.1.0 rc1, rc4  CC3D_OPBL through OpenPilot, Librepilot. Controller trying to start (long led signal), then after some time get led flashing serie, then led off and controller doesn't respond. 

Flashing 3.0.1 - all works fine.

Sheynar, 2017-01-08:
Rc5 still wont work (Same Led Code). Tried on Walkera runner 250c (cc3d) main controller, and more classik CC3D board. Both work fine on 3.0.1.

Sheynar, 2017-01-10:
Tried Rc6 CC3D_OPBL - same led code, still won't work.

S34nYun, 2017-01-10:
Betaflight 3.1 rc5, rc6 on CC3D does not work.
I can't  connect to CC3D via usb, I think that the CC3D board will does not start properly.
Flashing 3.0.1 - all works well too.

AndersHoglund, 2017-01-10:
Confirmed. 
Regular CC3D (non-OPBL) 3.1-RC6 does not connect to BFC. Tested USB and UART1/Main. 
CC3D_OPBL not tested.

AndersHoglund, 2017-01-10:
Seems target CC3D is broken from Jenkins build 911 onwards. Build 910 connects fine, both via USB and Main port.. 
iBus Telemetry was introduced here in commit 0169e943 .

When updating to current HEAD (7f86dee88adb0aeff6c09d8e99bbad0f7a25805f) and disable iBus Telemetry in the build, all is OK again. Added this to target/CC3D/target.h:

    #ifdef TELEMETRY_IBUS
    #undef TELEMETRY_IBUS
    #endif

Have not investigated deeper in the iBus Telemetry code what exactly is causing the problem. Marginal size differences though.
Sizes with iBus TM:

       text    data     bss     dec     hex filename
     119092    1380   15904  136376   214b8 ./obj/main/betaflight_CC3D.elf

Sizes without:

       text    data     bss     dec     hex filename
     119052    1380   15904  136336   21490 ./obj/main/betaflight_CC3D.elf

AndersHoglund, 2017-01-10:
Strange. With PR #2012 in Jenkins build 931 it all started working again. But OK, problem gone.

AndersHoglund, 2017-01-11:
Something weird is going on. CC3D with 3.1.0-RC7 fails to connect to BFC. Re-testing a few builds between RC6 and RC7 shows there are only a few working OK. Build 931, 932 and 933 are OK. Earlier or later builds fail.
And it seems it has nothing todo with TELEMETRY_IBUS. Still fails when turning this off now. Previous finding was probably a side effect of something else....

atomiclama, 2017-01-11:
I had some problems when I was doing local builds for my CC3D some time back.
A small change in code sometimes just reducing the size by a few bytes would cause the uP to abort during USB VCP startup. 
I was thinking along the lines of an alignment issue. 

But at some point after a git update it all started working fine. Never went back to it. 

Maybe this is happening again.

AndersHoglund, 2017-01-11:
Yes, must be something like that. Or some random content in un-initialized areas.
And now with build 937, it starts working again.... Weirdius Maximus.

bam0027, 2017-01-12:
For my clearer understanding, can you specify what color LEDs are flashing in your boot sequence?

When connected to USB (through a USB 2.0 hub), after flashing something recent (I'd have to research where I was at for specifics), my blue boot LED stopped working, and I only got a yellow LED that would blink moderately, and then slow to nothing.

Later, when I tried connecting to a 4A USB charger, and then directly to a 2S battery, I got the blue boot sequence again, but am still unable to get recognition through USB; tried cables, Mac and Win, even an FTDI interface. This is where I'm at now, with a board that can seemingly boot (based on blue LED sequence), but not connect via USB at all.

I'm just trying to see if my symptoms are related to this issue.

p.s. I was referred from [here](https://community.micro-motor-warehouse.com/t/betaflight-on-cc3d/1982/21?u=burtlo), because of [this](https://forum.librepilot.org/index.php?topic=2860.0) and [this](https://community.micro-motor-warehouse.com/t/betaflight-on-cc3d/1982/19?u=burtlo).

AndersHoglund, 2017-01-12:
Colours vary. I have a red power led and a green status led. Have no idea what your blue and yellow leds are. 
The power led is on as soon as power is applied. When everything works, there are a number of status led flashes. First a set of 5 quick flashes then a few slower, ending with 3 slow blinks indicating board is initialized and connectible. 
When failing I only see the first 5 quick flashes, then nothing more.

Tested builds 938 and 939 as well, all OK.

bam0027, 2017-01-14:
I'm going to lurk a bit more before getting into this, pardon. My issue wasn't involving BetaFlight per se, but I was experiencing a similar power issue, so I asked. I think you can pass by my issue, unless you think there's some way I can help; my board in question is just sitting, presumed "dead".

AndersHoglund, 2017-01-15:
I think I will give up on this one. 
Mainly because trying to run GDB on optimized code is a pain. Secondly when building a functionally stripped, non-optimized and debug-able version, the problem goes away. 
When trying to add trace-code, using LED blinks etc, it affects the code in ways that sometimes hides the problem and gives false clues. 
And last, it seems to work now with 3.1.0-RC8.   I suspect there still is something fishy though, just can't put my finger on it.

AndersHoglund, 2017-01-15:
Follow up, problem is back. Jenkins builds 946 and 947 fails, does not connect to BFC. 
Build 945 (RC8) still OK.

One additional clue: Build 947 works and connects to BFC using UART1/Main port, if MSP on VCP is turned off. (Used 945 to turn it off). I did also notice that there is no return from `mspSerialInit()` in `fc_init.c` when MSP on VCP is enabled. Could be some spurious interrupts when `openSerialPort()` is called from `mspSerialAllocatePorts()`. Have not followed up on that lead yet.

borisbstyle, 2017-01-15:
Cc3d really screams to be ditched!

AndersHoglund, 2017-01-15:
Yes. All F1 boards. We could leave them at 3.0.1
Unless there is a general VCP problem, only showing up on (some) CC3D builds...

mikeller, 2017-01-15:
Does this just affect CC3D_OPBL, or CC3D as well? Also, NAZE is F1 and still works find on 3.1...

AndersHoglund, 2017-01-15:
Have not tested CC3D_OPBL, only CC3D. 
I see there is a discussion on interrupts vs pass through on slack. This issue might also be a symptom of interrupts interfering during init.

Sheynar, 2017-01-15:
Tested CC3D_OPBL (on Walkera Runner 250c advance) have issues with startup (RC8 and some 1-2 builds before started). And i have issues with my motors/ESC something with mapping resources maybe: from flash to flash i have 5 or 6 reciever channels working, and one ore another motors responds to motor configurator, other 2-3 dont spin. And as symptom maybe that initiation of ESC got apart (three then 1 last, two then two etc.).

borisbstyle, 2017-01-15:
@AndersHoglund
NAZE doesnt really show any issues as far as I know.
CC3D has always been problematic

AndersHoglund, 2017-01-15:
OK, just thinking all F1s are really too small for BF now. Even if there are no catastrophic issues. 
With CC3D it seems as MSP on VCP is the current problem, as far as I can see. Only tested naked FC on the bench, no motors nor anything else.

atomiclama, 2017-01-16:
The problem is in the F1 USB VCP. 
Are there any other boards that use it?
Please don't ditch it yet.

AndersHoglund, 2017-01-16:
With PR #2134 there might still be some hope for the continued life of CC3D....

AndersHoglund, 2017-01-16:
Both targets CC3D and CC3D_OPBL tested with PR #2134 and they startup and connects to BFC properly over USB.

Sheynar, 2017-01-16:
Tested on Walkera Runner 250c commit [3c29725], starts, but ESC and motors works wierd. Initialisation of ESC got apart 2 motors, one motor, and some time later last motor, sometimes one of motors again. Something like on  [https://github.com/betaflight/betaflight/issues/2121](url). And there are some warnings on compilation process [[3c29725]_Build_Log_CC3D_OPBL.txt](https://github.com/betaflight/betaflight/files/709444/3c29725._Build_Log_CC3D_OPBL.txt)

AndersHoglund, 2017-01-16:
This sounds as a completely different issue, an ESC arming issue I would say. Check your ESC throttle parameters. 
The compile warnings are probably due to you using GCC 6.

AndersHoglund, 2017-01-17:
My CC3D runs 4 motors just fine on OneShot142. 
All settings default, except `min_command = 980`, as I have the ESC PPM Min/Max Throttle set to 1000 ... 2000.  BLHeli Passthrough also works. 

If you still see some issues, please open a new case, will close this one now.

Sheynar, 2017-01-17:
Rc9 same issue - CC3D don't startup, [b3c5980] build starts.

AndersHoglund, 2017-01-17:
CC3D RC9 works fine here.

    # version
    # Betaflight / CC3D 3.1.0 Jan 16 2017 / 23:11:05 (704b12f)

But you are right if you mean CC3D_OPBL RC9, that does not start. Darn.
Re-tested build 950, still works:

    # version
    # Betaflight / CC3D_OPBL 3.1.0 Jan 16 2017 / 11:24:00 (149ec6d)

Also, the CC3D_OPBL version you refer to, build 961 (b3c5980),  as working does not start up on my CC3D. But CC3D on the same code base do work:

    # version
    # Betaflight / CC3D 3.1.0 Jan 17 2017 / 14:46:12 (b3c5980)

So, there seems to be something wrong still. At least with the OPBL variant.

AndersHoglund, 2017-01-17:
Build 962, CC3D_OPBL does not start. But CC3D does:

    # version
    # Betaflight / CC3D 3.1.0 Jan 17 2017 / 23:31:19 (2e81bd0)

AndersHoglund, 2017-01-18:
@Sheynar could you please test the attached firmware, with PR #2153 included. 
[betaflight_3.1.0-PR2153.zip](https://github.com/betaflight/betaflight/files/714293/betaflight_3.1.0-PR2153.zip)

Sheynar, 2017-01-18:
Works (Starts up). Tested CC3D_OPBL on 2 different CC3D boards: Walkera Runner 250c, and more classic. Flashed through LibrePilot GCS.

AndersHoglund, 2017-01-18:
Great, thanx. 
I would also recommend you to get an USB-serial adapter and flash the native CC3D target via Main port using BFC. Much easier to handle connection to BFC without need for a separate power up. 
Leave the OP GCS world for good. Not sure how much longer we can keep CC3D_OPBL alive....

closed 2017-01-18 by AndersHoglund