betaflight/betaflight #2044: QX95 F3EVO ACC Trim backwards - BF3.1 RC4

dberrett, 2017-01-07:
Using the accelerometer trims on the config page (or using the "in flight" adjustments), the trim seems backwards.  i.e. If I am drifting right or forward, I have to set positive trim (which is different than previous releases).

rodgerramjet, 2017-01-27:
I can confirm this bug on SPRACINGF3EVO 

Its been a while since I used ACC so not sure when this bug happened, but it was the other way round in CF 1.13 which I just upgraded from (on a brushless micro) and doesn't really make logical sense when you are using TX commands to set adjustments (e.g. if I'm quad is drifting left I have to go throttle HIGH and roll LEFT - odd to adjust in the direction of problem)

Eganwp, 2017-01-27:
Haha I also noticed this bug on my X-Racer SPI board... however since I'd never tried the feature before I wasn't positive it was a bug. It is annoying having to trim in the direction the quad drifts... completely backwards to how we normally trim. :)

Eganwp, 2017-01-27:
To clarify, this bug was on RC13 on my X-RACERSPI FC.

BenniUK, 2017-02-09:
accel trimming is reversed on the AIRBOTf4 3.1.5 also