betaflight/betaflight #2047: CPU% higher & LEDs causing issues

Aileron8, 2017-01-08:
The CPU% appears to have increased on 3.1.0 RC5 since updating my BFF3 & XRacer boards. Unchanged for my Kombinis, but I hadn't updated those in over a month. Where as my BFF3 had the December 14th stock fw and my XRacers build#823 before tonight's updating. My BFF3 had DShot600 loaded and was working alright until I enabled the LEDs on the configurations toggle. When I reconnected the orientation was off and I was unable rectify using the board and sensor alignment functionality in the configurator. I had to reflash and leave the LEDs off, but leaving DShot600 was alright. I successfully flew with DShot600 or MS loaded onto a Revo Clone, Kombini, and four XRacerF303's without issues other than the increased CPU% rates.

borisbstyle, 2017-01-08:
On my bff3 i had 35% cpu with led and dshot600 enabled. Just tested that like 10hours ago

Aileron8, 2017-01-09:
I'm not sure why I experienced the LED/DShot conflict with my BFF3, but after reloading RC5 yesterday using my Mac it is functioning properly. The CPU% is still higher than before, but I am pleased with BF3.1.0 RC5 overall. Thank you

borisbstyle, 2017-01-09:
RC6 will have much less cpu again ;)

closed 2017-01-09 by borisbstyle