betaflight/betaflight #2049: Enabling DISPLAY on Omnibus target locks up the FC

MallyAzure, 2017-01-08:
On betaflight 3.1 RC5, enabling the Display feature locks up the fc and can only be recovered by reflashing it.

mikeller, 2017-01-08:
Locks up on a clean install on AIORACERF3 as well.

jflyper, 2017-01-09:
@MallyCanFly @mikeller Did you have OLED connected?

Few observations

- It does not "lock up" if I2C OLED is actually connected (OMNIBUS)
- It is not "locking up" without the OLED. Wait with patience, it will be responsive in 10 sec or so; it's probably trying to find the nonexistent OLED. (1000+ i2c errors when it finally comes up.)
- It is not RC5 specific. Has been there for a long time.

Pragmatic solution
- Don't turn on DISPLAY without the OLED.

olexs, 2017-01-09:
Can confirm identical behaviour as described by @jflyper on a SPRACINGF3 board, with both RC4 and RC5.

mikeller, 2017-01-09:
@jflyper My test was without a display connected. And you are correct, the board does come up after ~10 seconds.

Unless there is a way to cut back on the time until giving up (making 1000+ attempts before giving up sounds excessive), this looks like a documentation issue, which can be solved with a tooltip in the configurator. (But still a suboptimal solution, thinking of users that have a display connected that isn't powered when running on USB...)

jflyper, 2017-01-09:
It should produce one error at the most per page cycle of the dashboard... @martinbudden any thoughts?

martinbudden, 2017-01-09:
@jflyper , @digitalentity has done a lot of work one the I2C drivers in iNav, so I'd say the best course is to say this is a known problem and do a proper fix for it in 3.2

jflyper, 2017-01-09:
@martinbudden Yeah, I saw the PR. I like the clean code and also robust and quick recovery from errors.

MallyCanFly, 2017-01-09:
@jflyper Can confirm the behavior you described with the omnibus.
What I would use the display for is as a field tuning tool that I can disconnect when not in use, so this issue would make it a bit cumbersome.

jflyper, 2017-01-09:
@MallyCanFly 
I understand. The authentic way of using the OLED display.

@martinbudden 
1. It's caused by `dashboardInit` calling `displayPortOledInit` without checking `dashboardPresent` flag, which eventually calls `i2c_OLED_clear_display_quick` that produces the massive errors.
Quick and incomplete fix is to check `dashboardPresent` before calling `displayPortOledInit`.
This fix does not consider OLED connected after the `dashboardInit` is called from `main`...

2. I also realized the code does not handle OLED disconnected.

3. Another quick fix would be to create and check liveness (existence) of the device at driver level.

martinbudden, 2017-01-09:
@jflyper , I'll do a PR for (1). We can revisit this once 3.2 has been released.

martinbudden, 2017-01-09:
OK, I've had a look at the code and it seem that there is an assumption at a fairly fundamental level that the user won't turn on the dashboard unless there is one connected. So a "quick fix" risks breaking this. So it's too late to make a change for this for 3.1. We'll revisit for 3.2. As @jflyper states, this is not something that has recently been broken, rather it is the behaviour we have had for some time.

mikeller, 2017-01-09:
Fair enough, I'll add a tooltip explaining this to configurator.

mikeller, 2018-02-01:
Closing as fixed (by documentation).

closed 2018-02-01 by mikeller