betaflight/betaflight #2050: 3.1.0-RC5 SPF3EVO SUMD mode not working

mhunley123, 2017-01-08:
Setup with SerialRX on UART2 using serial-based receiver mode and SUMD on receiver provider.  ESC/Motors setup as PWM.  All channels in the receiver tab on the configurator(1.8.5) change and operate as normal when moving transmitter controls.  On actual quad, all aux channels work, motors arm using aux1, but roll/pitch/yaw/throttle do not work.  I switched back to 3.0.1 with same setup and everything works fine.

mikeller, 2017-01-08:
Have you double - checked the PID tab to make sure your rates aren't set to 0?

If that doesn't solve it, can you please post a CLI `diff`?

mhunley123, 2017-01-08:
Hoping the rates would be a quick fix, but they were not 0...  here are the diff's from both setups.

[3.0.1.txt](https://github.com/betaflight/betaflight/files/692469/3.0.1.txt)
[3.1.0.txt](https://github.com/betaflight/betaflight/files/692470/3.1.0.txt)

TheAngularity, 2017-01-08:
`set pid_process_denom = 16` in 3.1.0 is wrong ... set it to defaults

mikeller, 2017-01-08:
@TheAngularity There is nothing wrong with `pid_process_denom = 16`. Starting with 3.1.0 the range is 1 - 16 for `pid_process_denom`.

TheAngularity, 2017-01-08:
@mikeller 
I just tested it and it does not work

mikeller, 2017-01-08:
@TheAngularity What did you test?

TheAngularity, 2017-01-08:
what are we talking about right now? ... @mhunley123 3.1.0 diff's works with 8k gyro update and a pid loop with 0.5k ... this i have tested on my evo and it's not running ... motors armed but throttle slows the motors and nick/poll/pitch not working

mhunley123, 2017-01-08:
I just changed the pid_process_denom to 5 in 3.1.0 and it works now.  
The only setup that works is;
Gyro update at 2.67KHz
PID loop at .53KHz

closed 2017-01-09 by mhunley123