betaflight/betaflight #2054: Transponder/Kombini/I-Lapse

Gripchion, 2017-01-09:
You can beat me up if I'm not supposed to ask this here. If I have done everything correctly should the Race Transponder feature in Betaflight be working with a Kombini and an I-Lapse system or does it still need to be integrated. I have a done a bunch of research on the forums and asked questions of Furious FPV. No answers. Now I am here. Thanks much in advance.

jflyper, 2017-01-09:
@Gripchion Can you elaborate? Which release? What is the nature of the problem you are having?

Gripchion, 2017-01-09:
@jflyper I have a Kombini FC and an I-lapse System. Using Betaflight version 3.0.1, configuator 1.8.5. I can not get the Kombini FC Transponder to talk to my I-Lapse system. What I have done so far is. 
1. Installed IR LED with flat side notch lead to the trans -, the other lead is to trans +.
2. Activated Expert Mode and enabled Transponder Tab in Betaflight.
3. As per Furious FPV Video I have copied and pasted the last 12 digits of the Kombini Unique Device ID to the Transponder ID Box in Betaflight. Then saved. Epprom shows saved. 
4. Disconnected FC from Betaflight, power up FC. Checked IR Led with camera on phone. IR Led is pulsing.
5. Power up I-Lapse go to Edit Racer Tab, Click on Create New Racer. Click on Scan. Program shows scanning, but the program/recievers will not pick up the signal from the IR LED on the Kombini. 
6. Confirmed that the I-lapse system is working with 2 other transponder bought from I-Lapse. In the Scan Mode. The transponder from I-lapse are a stand alone unit.
7. Confirmed that the I-lapse system is not registering the IR Led signal from 2 other Kombini FC's that a friend has and has wired the same way as stated above.
8. Take the Last 7 digits of the Unique Device ID of the Kombini and added 00000 the front and end, saved those in the Transponder box in Betaflight, that didn't work.
9. Tried the first 7 and 12 digits of the Unique Device ID of the Kombini, no luck there.
I don't think I missed anything on my discription of my troubleshooting. The Transponder Code that is entered in Betaflight has to be 12 characters or it will not save. The UID for I-Lapse is only 7 characters????  When it comes to programing I don't know jack, but I love to troubleshoot. I'm at a loss. Thanks for trying to help me out. I appreciate it very much.
Jay

Gripchion, 2017-01-18:
Updated to Gui 1.8.9, BetaFlight 3.1, recieved transponder I lap codes from Furious FPV. With Betaflight 3.1 the transponder IR Led will not flash. Didn't change any wiring, IR LED was flashing as described above.
J