betaflight/betaflight #2055: Spektrum Sat SPRF3 Issue

pigd0g, 2017-01-09:
RC5 Issue with SPRF3 Spektrum Sat (Lemon Diversity).  Connected to UART3. Works fine in 3.0.1. In 3.1-RC5 board times out receiving data and cannot re-connect (appears to work okay until RC protocol is selected), have to disconnect sat in order to re-flash 3.0.1 otherwise cannot contact bootloader.

Diff from working 3.0.1 for reference:

# name
name 210

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature -BLACKBOX
feature RX_SERIAL
feature TELEMETRY
feature AIRMODE

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1800 1900
aux 1 12 0 950 1025
aux 2 12 0 1650 1725
aux 3 12 0 1275 1350

# adjrange

# rxrange
rxrange 0 1140 1857
rxrange 1 1150 1857
rxrange 2 1145 1857
rxrange 3 1141 1857

# rxfail

# master
set roll_yaw_cam_mix_degrees = 15
set min_throttle = 1046
set motor_pwm_protocol = MULTISHOT
set small_angle = 180
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 32
set align_board_yaw = 270
set gyro_sync_denom = 2
set yaw_deadband = 15
set acc_hardware = NONE
set pid_process_denom = 1

# profile
profile 0

set dterm_setpoint_weight = 168
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 129
set rc_rate_yaw = 120
set roll_srate = 79
set pitch_srate = 79
set yaw_srate = 76

#

blckmn, 2017-01-09:
Can you please try turning off telemetry? And let us know if that resolves it. If so I'll have a fix prepared for next RC.

pigd0g, 2017-01-09:
Good work @blckmn that resolves the issue...

blckmn, 2017-01-09:
So you're not using telemetry at all?

pigd0g, 2017-01-09:
I just pasted in the above diff without the telemetry option set and it behaved normally. I use telemetry to output to a micro minim osd on the other UART

Arakon, 2017-01-09:
MinimOSD doesn't even use telemetry, OSD uses MSP.

pigd0g, 2017-01-09:
@Arakon Thanks for confirming, I was reading up on MSP in the scarab osd thread.  I guess there's still a conflict on the board between the telemetry and spektrum sat though.  I'll connect up the cam etc and go for a fly tonight.

AndersHoglund, 2017-01-10:
As Spektrum SRXL Telemetry now (partly) being available in BF, there is a small "gotcha". SRXL requires a bi-directional Satellite data line. Normally, without TM, Satellite is connected to UART uni-directional Rx line. When Telemetry is turned on, the UART Tx line is used for bi-directional Rx/Tx satellite data. So, you have to connect you Sat to either Rx or Tx depending on TM selection. Very inconvenient and will create many support issues and Q's.

I did try to duplicate the reported issue by turning TM on and off on my SPRF3 with a normal (non-TM) Satellite connected to UART3 Rx, and I could not see any bad effect on USB serial connection to BFC. Connected just fine as usual. But there might be more to it to be able to reproduce this problem, just made a quick check.

Ref related PRs: #1798 , #2012 and #1766    A bit of unfinished business.

blckmn, 2017-01-10:
@AndersHoglund yes -a bit of an oversight on my part. If someone enables telem - it might have been for another protocol.

I've tidied this up and a few other items in this PR: https://github.com/betaflight/betaflight/pull/2012

Just have a few build errors to sort

pigd0g, 2017-01-10:
@AndersHoglund great explanation.  Yes I'm sure this will cause a few people to get stuck.  Close the issue, but maybe update the spektrum sat notes somewhere :+1:

AndersHoglund, 2017-01-10:
Thnx. Yes, the Spektrum TM SRXL do really need some user documentation. Not sure where and how. The average user don't read docs anyhow, they connect, power up and ask/complain. Things need to be automated, "PnP". Not possible in this case from what I understand. Not on all cpu types at least. Might be doable on F3s, keeping bi-directional data on the Rx line. But still, it is better to have all targets behave the same. "Special cases", some do this, some do that, only complicates things even more.

closed 2017-01-10 by AndersHoglund

blckmn, 2017-01-10:
I'll add something to the wiki, but it's still likely people won't find it / won't read it.

SpektrumFPV, 2017-01-10:
Yes bi directional SRXL requires connection to the TX pin. I believe F3's allow interchanging the TX and RX pin in firmware though, so it would be possible to just configure the RX  pin to be TX pin, and set it in half duplex to allow receiving. This could be the case for any of the receiver protocols since you can configure the TX pin in half duplex to be able to receive, without ever actually requiring it to transmit back. This way, people wouldn't have to read up on special connection usage depending on whether they're using a bi-directional receiver or not. It would still be connected to the rx pin, at least on F3's. Firmware pin interchanging for UARTS isn't possible on F4s or F1s.