betaflight/betaflight #2061: CMS over SmartPort

olexs, 2017-01-09:
As discussed in #1311, this is intended for discussion and development of a full "CMS over SmartPort" solution. The idea is to combine the existing `displayport_msp` and MSP-over-SmartPort mechanisms to send CMS messages down to the transmitter, and render the CMS menu on the transmitter display with a Lua script. Following steps are necessary, as far as I understand:

- Current MSP-over-SmartPort implementation can only send MSP packets as a response to requests from the client (transmitter). A mechanism to send spontaneous, FC-generated packets (specifically MSP_DISPLAYPORT messages) is needed.

- Re-transmission mechanism for CMS-over-MSP.

- Lua code for MSP parsing needs to be extended to correctly receive and decode MSP_DISPLAYPORT messages.

- Lua code needs to be written to actually render MSP_DISPLAYPORT contents on the Taranis screen.

raphaelcoeffic, 2017-01-09:
@olexs Thx for this sum up!

I'd add another item, which is that we might need a mechanism to relay the key-presses on the TX, rather than using the sticks to control CMS.

DieHertz, 2017-01-09:
Looks exciting, too bad Er9X does not support any of this goodness.

raphaelcoeffic, 2017-01-09:
@DieHertz Maybe time for you to get yourself a proper OpenTx-compatible TX :-D

DieHertz, 2017-01-09:
More like "time to finally work on opentx/opentx#3200" :smile:

olexs, 2017-01-09:
@raphaelcoeffic as a future expansion, sure. For a start though, simply rendering the menu in the exact same way as MW-OSD does it would be enough to make the CMS usable without either an OSD or an OLED display.

Disclaimer: now that SmartAudio is available, I have a dire need of this, to control my Unify Pro HV hidden under an aero pod without having to install an OSD :)

raphaelcoeffic, 2017-01-09:
@olexs I perfectly understand the background and use of all this! I know some other guys not using any OSD, or willing to be able to configure things with the VTX switched off (for example at racing events).

raphaelcoeffic, 2017-01-09:
concerning the VTX configuration itself, #2018 might be an option as well ;-)

olexs, 2017-01-09:
True, #2018 might be a good alternative for VTX configuration that does not depend on CMS.

Still, having all the CMS menus on the Taranis would make many things much easier and quicker in the field, largely eliminating the need to bring and connect a laptop to setup and tune a new copter.

How do we want to approach this? As I read #1311, you wanted to start working in this direction back in November when CMS changes were merged. I'd like to contribute (as far as my limited C and Lua knowledge allows), but would prefer to avoid change conflicts if possible.

raphaelcoeffic, 2017-01-09:
@olexs No worry, it's next on my list! I currently have a Horus for 10 days, so that I need to port the LUA scripts there first.

olexs, 2017-01-09:
@raphaelcoeffic Alright :) If there is anything I can help out with, let me know.

closed 2017-01-31 by borisbstyle