betaflight/betaflight #2063: Smartport telemetry on F4 (mainly SOULF4)

mickkn, 2017-01-09:
Okay, sorry for making this issue, if it's a known issue. A confirmation that it is not working properly yet would be fine :)

I've got a Demon Soul F4 Flight Controller and Smartport telemetry is not working.

I've hooked the s.port input (the connector) from the X4R-SB to the labeled S.Port output on the FC. The S.Port output is UART6(TX) I assume, and is hardware inverted on the flight controller I assume (since they write: "By using built in Smartport Inverter Telemetry, you have full control of your lipo voltage.")

http://demonrc.eu/product/demon-soul-f4-high-performance-flight-controller/

As far as I know this goes:

X4R-SB Smartport (Connector): Inverted input
Demon Soul UART6(TX): Inverted ouput

So it SHOULD work ? :)

A solution I could think off but haven't tried yet could be soldering a cable before the inverter on X4R-SB and to the un-inverted UART3(TX), would that work?

My cli diff dump:

# diff

# version
# Betaflight / SOULF4 3.1.0 Jan  4 2017 / 13:43:48 (eb87c1c)

# name
name MixukoF4

# resources

# mixer

# servo

# servo mix

# feature
feature VBAT
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 5 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 12 0 1300 1700
aux 2 25 2 1700 2100

# adjrange

# rxrange

# rxfail

# master
set debug_mode = NOTCH
set min_throttle = 1065
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = SBUS
set telemetry_inversion = OFF
set pid_values_as_telemetry = ON
set current_meter_offset = 10
set current_meter_type = VIRTUAL
set acc_hardware = NONE
set pid_process_denom = 1
set blackbox_rate_denom = 4
set blackbox_device = SPIFLASH

# profile
profile 0

set vbat_pid_compensation = ON
set p_pitch = 63
set i_pitch = 53
set d_pitch = 27
set p_roll = 63
set i_roll = 53
set d_roll = 27
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 110
set roll_srate = 75
set pitch_srate = 75

borisbstyle, 2017-01-09:
Sorry i will have to close this one. You will have to contact the shop to ask for installation detaills.

Telemetry works for sure, but it needs inverter or you can bypass the inverter on your receiver.

closed 2017-01-09 by borisbstyle

mickkn, 2017-01-09:
Thanks for the reply boris

borisbstyle, 2017-01-09:
The inverter is something additional. Its not driven by the firmware. Its a piece of hardware on the board.
We do get these questions a lot from f4 users.

mickkn, 2017-01-09:
Yes, the telemetry_inversion doesn't do anything on the F4 boards right?

borisbstyle, 2017-01-09:
Nope....only works on F3 hardware

mickkn, 2017-01-09:
I'm just trying to check things off :) now I know the software should work. But if the FC do have inversion on the UART6(TX) it "should" work when connected to the X4R-SB connector

DieHertz, 2017-01-09:
@mickkn I will investigate this once I get home in a week, I have some SOULF4 boards to look at. Do not lose hope :)

mickkn, 2017-01-09:
Thanks DieHertz :)

mickkn, 2017-01-09:
I've done the non-inversion mod from https://blck.mn/2016/06/smartport-the-frsky-x4rs-and-betaflight/ and it works on UART3. Guess their inversion circuit is not working.

mikeller, 2017-01-09:
@mickkn: It could also be that the SOULF4 has got another switchable external inverter on UART 3 (like https://github.com/betaflight/betaflight/blob/master/src/main/target/REVO/target.h#L72-L73 for UART1), but it's not defined in the board definition. I think @DieHertz has got contacts at the manufacturer.

mickkn, 2017-01-09:
So there could be a software issue?

mikeller, 2017-01-09:
@mickkn: Looking at the board pinout on their website more like a misunderstanding. It looks like they put the circuitry for SmartPort onto UART6, not UART3. Therefore I would expect UART3 to be not inverted.

mickkn, 2017-01-09:
Yes it's on UART6 but is not working, so I took the non-inverted input on X4R-SB and connected it to the UART3 which is non-inverted. This solution works. Using the connector on X4R-SB and the UART6 does not work

mickkn, 2017-01-11:
@DieHertz did you do some investigations? :)

DieHertz, 2017-01-11:
Not yet, I'll return home to my souls this weekend!

mickkn, 2017-01-15:
@mikeller did the last commit with serial inverter support fix this issue ? :)

mikeller, 2017-01-15:
@mickkn I wouldn't think so, since it doesn't change anything for existing targets. It just adds support for future targets to have more than one external inverter.

Matthias7Song, 2017-02-16:
@mickkn DemonRC's Facebook page says that S.Port telemetry issue has solved;

Finally Smart Port telemetry works on BF. All you have to do is to type these 2 commands in CLI and save:
set tlm_inversion = OFF
set sport_halfduplex = OFF

I connected S.Port wire to S.Port pinhall on the board and set Smartport on UART_6 in Betaflight configurator. Then once I put in those two commands, it works.

piepluquad, 2017-08-24:
maybe this is an answer : frsky inverters.
https://www.youtube.com/watch?v=gWg9R7OLe5c

piepluquad, 2017-08-24:
maybe this is an answer : frsky inverters (does apply to XSR-E too)
https://www.youtube.com/watch?v=gWg9R7OLe5c