betaflight/betaflight #2066: Betaflight 3.1.0 RC5 GPS is partially broken on SPRACINGF3

nhadrian, 2017-01-09:
Hi,

I have just uploaded the RC5. It looks like there is some kind of issue with GPS, after enabling GPS, the icon on the Configurator blinks. So either a configurator or a Betaflight issue. 

In addition, the GPS datas are not forwarded through HOTT telemetry, only other values like current, etc.
But my transmitter says GPS module is connected, so HOTT GPS emulation works, only values are not forwarded.
 
I'm using an MTK3329 GPS in NMEA mode with 112500 baud, GPS auto config is on, so the receiver is configured via Betaflight.

Could anybody test GPS functionality on F3 targets?
Thanks in advance.

Regards
Adrian

nhadrian, 2017-01-09:
I double checked connections so I'm quite sure GPS connection is OK. (BTW I didn't touched wiring during update from 3.0.1 to 3.1.0 RC5).
Here is a short video of the issue: https://www.youtube.com/watch?v=BQ9ZhfPmZcA

borisbstyle, 2017-01-09:
Gps is only enabled on f4 targets on 3.1 for now.

It had to do with flash size hitting limit

closed 2017-01-09 by borisbstyle

wind0r, 2017-01-09:
#2022 should have reenable gps for most f3 targets

also @nhadrian wrote that gps is working sometimes.
Since i do not have a gps device i cant debug and help here :(

nhadrian, 2017-01-09:
Yes @wind0r you have right. #2022 reenables GPS on F3 targets. 
If you look at the video, some datas are read from GPS, but perhaps there is something with the NMEA parser or the schedule of NMEA parsing because it looks like GPS is recognized again and again once valid data is parsed...