betaflight/betaflight #2069: DSHOT not working on Youbi XV130  SPRF3/F_H_40

glenb2, 2017-01-09:
Just getting random initialization beeps and sporadic motor speeds

mikeller, 2017-01-09:
Did you remove the filter caps from the ESCs?

glenb2, 2017-01-09:
I thought that only had to be done for DSHOT600?  I'm cool with 150 or 300 and am getting this issue on both.  Do you have to remove filter caps for 150 and 300 too?

blckmn, 2017-01-11:
That depends entirely on the hardware. Given it is not working for you, it is either the ESC or the FC. If you're comfortable and capable then remove the caps and try again.

chubbs456, 2017-01-17:
Dshot150 worked on  my xv130. Motors were all synched and smooth at 1010. Although it seems air mode was always on and motors would always spin no matter the idle setting.

glenb2, 2017-01-18:
OK, I'll try RC9 and follow up

Cheredanine, 2017-01-19:
@glenb2 
The dshot section of the wiki lists stuff that has been reported in the rcgroups thread.  But that is the Internet....
Slightly different hw can produce different results for example I have 2 quads with Aikon semf 30a v1 escs. One uses a bluejay f4 flight controller and flies fine with dshot 150 or 300 with the caps still on the escs. The other uses a dtfc (doge target) flight controller and will not run any dshot without the caps removed.

So although that section of he wiki can be used as a guide, if dshot doesn't work and you still have caps on, the general rule is take the caps off.
If you still have a problem the. There may be an issue

nathantsoi, 2017-04-25:
@glenb2, did it work? mind closing this issue if you're good?

closed 2018-04-21 by DieHertz