betaflight/betaflight #2077: What have i done? Please Help?

Jdog-glossop, 2017-01-10:
I've bought a Flycolor Raptor to put in a 130mm build
It's a SPR F3 flight controller and I connected to cleanflight and betaflight went to flash it to the latest betaflight and it bricked went to bootload it and that didn't work... So then started watching videos on youtube one with Joshua bardwell and then another one and being an idiot I went on to zadig and changed the driver? (I think) on the board and now betaflight cant find it ? nothing show to connect to..
CAN ANYBODY HELP PLEASE ? The light still come on ! and my other quads have no issues making me think it the flight controller 
Cheers Jamie

robert-hoffmann, 2017-01-10:
Give this a try
http://www.dronetrest.com/t/fix-any-stm32-dfu-drivers-issues-when-flashing-betaflight-cleanflight-firmware/3603

I had problems getting my board into DFU mode ..this worked well for me

Jdog-glossop, 2017-01-10:
Cheers Robert
sorted it ;]

closed 2017-01-11 by blckmn

robert-hoffmann, 2017-01-11:
Cool, glad to hear !