betaflight/betaflight #2079: revo 3.1 beeper

Jama85, 2017-01-10:
Hello

On the revo board the beeper constantly beeping, before with 3.0 working well.

blckmn, 2017-01-10:
It's likely you don't have a Revo, but rather an AirbotF4.

blckmn, 2017-01-10:
The Revo doesn't have a built in beeper.

Jama85, 2017-01-10:
its not a built in beeper.
http://www.banggood.com/Raceflight-Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-with-VbatBuzzer-p-1087561.html

mikeller, 2017-01-10:
Have you tried flashing the AIRBOTF4 target?

Jama85, 2017-01-10:
Still not, but before its worked with revo firm.

mikeller, 2017-01-10:
I don't think AIRBOTF4 was a separate target in 3.0, but now it is. Using the incorrect firmware for the hardware that you have will cause it to not work correctly.

blckmn, 2017-01-10:
ok try the following set commands:

- beeper_inversion
- beeper_od

BUT this board is definitely a AIRBOTF4.

http://www.banggood.com/Raceflight-Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-with-VbatBuzzer-p-1087561.html

[image]

atomiclama, 2017-01-11:
That's the board I use and it works fine with AIRBOTF4 build from a week or so back.
Need to test a RC though.

Jama85, 2017-01-11:
Ahh ok, then i install airbotf4.

blckmn, 2017-01-11:
Yes. Please reopen if it remains an issue.

closed 2017-01-11 by blckmn