betaflight/betaflight #2080: OSD flicker

Gaz-B, 2017-01-10:
On all builds I have tried including #930 there is a visible flicker to the osd graphics viewed live I believe they are only being displayed on alternate fields as on the DVR recording Skyzone built in recorder there are no osd graphics recorded so I believe to save space the goggles are only recording one field also which happens to be the non osd one. this is on Kiwi F4, Foxeer Monster cam, ImmersionRC Tramp VTX.

jflyper, 2017-01-10:
Can you summarize the issue you have with betaflight firmware?

Gaz-B, 2017-01-10:
Lets try the OSD flickers very fast and looks transparent viewed live and when I record to my goggles it's completely missing.

borisbstyle, 2017-01-10:
This totally sounds like you didnt connect the gnd properly.
This is not fw issue

Gaz-B, 2017-01-10:
ssing Ok if you believe so I will pursue that route seems a bit odd that the video passing through the board is perfect. The FC has other issues no current sensing from new so a rebuild will be soon replacement is in the post will belt and braces grouns then thanks

Gaz-B, 2017-01-13:
New Kiwi fitted all grounds checked and fine. flicker visible PAL on skyzone goggles and blackpearl screen and as mentioned before the flickering OSD is not recorded at all using Skyzone dvr when using laforge receiver all looks ok no flicker and could record OSD. In NTSC everything works as it should no flicker on anything and can record OSD on skyzones I have tried setting OSD video standard in config to both PAL NTSC and auto and makes no difference I wonder if the OSD is permanently NTSC.

jflyper, 2017-01-14:
@Gaz-B 
So, is this a different issue?

Gaz-B, 2017-01-14:
Changing the OSD TV system in the configurator has no effect the only way to stop the flicker is to use an NTSC Camera. I looks to me that the OSD is fixed in NTSC.

jflyper, 2017-01-14:
> Changing the OSD TV system in the configurator has no effect the only way to stop the flicker is to use an NTSC Camera.

It the flickering CAM is 16:9 by any chance?

>I looks to me that the OSD is fixed in NTSC.

Let me look into this.

Gaz-B, 2017-01-14:
Yes it's a foxeer monster.

jflyper, 2017-01-14:
As far as I've seen in many threads in RCG and fpvlab, MAX/AB7456 has compatibility problem with 16:9 CAMs. You said the FOXEER Monster has no problem with NTSC? What is the native mode of the FOXEER Monster? NTSC?

Gaz-B, 2017-01-14:
I don't know if it has a native system the thing that seems odd is changing between NTSC ans PAL in the BF OSD configurator has no effect.

jflyper, 2017-01-14:
NTSC and PAL setting on FOXEER MONSTER seems to be switchable with CAM's native OSD menu.

BF OSD configuration, I'm still looking. I might have added forced auto detect mode...

Gaz-B, 2017-01-14:
I might have misunderstood your question the cam is switchable via built in osd that's how I am switching TV systems to test BF OSD on both.

jflyper, 2017-01-15:
@Gaz-B 
Can you check if OSD setting of NTSC/PAL is actually fixed at NTSC by not connecting CAM but only a VTX (display device direct is preferred)? PAL should bring slightly larger screen.

Gaz-B, 2017-01-15:
with no cam I can change systems the blackpearl doesn't auto select so if wrong picture rolls. I have to do a complete power reset battery and usb for changes to take affect .
NTSC CAM NTSC OSD no flicker 
NTSC CAM PAL OSD   flicker
PAL CAM PAL OSD      flicker
PAL CAM NTSC OSD   flicker

rhinorap, 2017-01-16:
If the code for the OSD is ported from the earlier (somewhat flaky) multiWii osd code then there has been an underlying fault with the code and some camera combos for years - and no one has managed (or bothered trying) to track it down. These symptoms also manifest on the microminim OSD AND the minimosd variants with some camera combos.

Gaz-B, 2017-01-16:
NTSC it is then my goggles are 480 line so it's a gain in frame rate rather than a loss of resolution. I thought it might be an issue you needed to be aware of to improve the product for all

closed 2017-01-16 by Gaz-B

l2smith2, 2017-02-20:
I discovered this problem today also. OSD flickering on Pal 16:9 hs1189 camera. Using a 4:3 hs1190 Pal there is no flicker. Skyzone goggles don't record the OSD

jflyper, 2017-02-20:
@l2smith2 Yes. Hardware compatibility problem.

Gaz-B, 2017-02-20:
Set the 1189 to ntsc using mine with 16/9 skyzones it like flying in HD

jflyper, 2017-02-23:
@Gaz-B 
HS1189 with NTSC 16:9 works with Skyzone goggles?
How about overlay recording?

Gaz-B, 2017-02-23:
Yep ntsc is perfect. In pal there was nothing in playback.

8bitreid, 2017-04-08:
I am getting this issue as well. When I apply any stock movement my osd flickers and is almost transparent. I will switch to NTSC and see if that fixes it. I have latest FW. Betaflight F3. Unify race pro HV. I will check the ground as well.

martinsv, 2017-04-16:
Had the same issue with foxeer monster v1 and PAL. Switching to NTSC solved, but that's not right solution :)

joseico90, 2017-08-31:
-Racerstar StarF3S AIO
-Skyzone V2 16:9 goggles
-Monster 16:9 camera
-HS1177 Camera (also tested)

*NTSC CAM NTSC OSD= Okay on ground.  Garbled picture once flying , unfliable video, like on another close by channel.
*PAL CAM PAL OSD =flicker

Gaz-B, 2017-08-31:
I now use ntsc on my odd and camera the horizontal lines match the goggles anyway so just getting an extra 5 frames per second. Worth trying worked for me.