betaflight/betaflight #2081: F3EVO DSHOT motor1 turning off when armed

mhunley123, 2017-01-10:
I just installed Favourite LittleBee-30A-S ESC BLHeli_S (removed cap's) on my F3EVO using 3.1.0-RC6.  Updated them using pass through to version 16.6.  Using Dshot 300.  Motors 1,2,3 spin and 4 does not, as expected.  I then connected motor 4 and 5 with wire as described.  Followed wiki to remove motor 5, move 4 to A06 and save.  Using motor tab in configurator, all 4 motors spin up great as expected.  I then arm the copter and all 4 motors spin, but motor1 quits immediately.  2,3, and 4 continue just fine.  I then go to motor tab in configurator and I now have no control of motor1 anymore.  If I reset the board, I get motor1 control back in motor tab of configurator, until I arm the copter.  I'm using serial SUMD on UART2.  If I configure motor4 back to A03, motor4 stops working, but motor1 works just fine.
[ResourceList.txt](https://github.com/betaflight/betaflight/files/697124/ResourceList.txt)

blckmn, 2017-01-10:
Do you have BB enabled on arm?

Try turning it off.

mhunley123, 2017-01-10:
No.  Its enable with an aux channel.  Here is the dump file if that helps.
[Dump.txt](https://github.com/betaflight/betaflight/files/697299/Dump.txt)

mhunley123, 2017-01-10:
I just got it working, but not sure why...  I started changing DSHOT, Gyro frequency, turning off volt meter, accelerometer, blackbox, and OSD, and changed blackbox from 4khz to 1khz.   It started working at that point.  So I then added everything back to where I started and it still works...

mikeller, 2017-01-10:
@mhunley123: Could it be that you had the switch on 'enable blackbox' accidentally when you tried before? Does it still work when you enable blackbox logging?

mhunley123, 2017-01-10:
yes, it still works with blackbox enabled.  It seems like it started working after i hit the "save and reboot" button on the blackbox configuration tab.

closed 2017-01-11 by mhunley123