betaflight/betaflight #2088: Updates for Dshot wiki

Peter010103, 2017-01-11:
Hello,

I have tested the Racerstar V1 20A ESCs with Dshot 300 from one of the 3.1 alphas. I can say that they are able to run the protocol... I am using the LUX v1 FC.
----------
Could we have an updated list of boards that support Dshot? I know RF1 will support dshot for the Revolt flight controller, but there was no mention on the wiki here whether 3.1 will support it.

blckmn, 2017-01-11:
You can update the wiki here: https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol

Keeping it up to date is similar to the code base, there is a reliance on those who successfully test various combinations to put those details in the wiki page.

It probably needs a bit of a clean up, and re-ordering (perhaps splitting the page across a few pages)

winkingsl, 2017-01-17:
Hello! 
I have tested Racerstar 4in1 20A V1 (cap removed, Firmware 16.5) with Dshot 300 on F4 REVO clone with 3.1.0 RC7. its able running smooth only with 8k/4k and  4k/4k (may be lower too). With 8k/8k/Dshot 300 and 8k/4k/Dshot 600 motors was twitching on point >=1600 from motors tab in configurator. 

Racerstar RS30A LiteS (Firmware 16.5) running smooth with 8k/8k/Dshot 600 without cap removed on same FC.

Maybe this information will be useful.

P.S. Sorry for my english.

Peter010103, 2017-01-17:
Your English is fine :)
Dshot 300 works fine on the RS V1 20A without the cap removed, but just to be safe, do you have any photos for which cap to remove? I've added this ESC to the DShot list on the wiki.

atomiclama, 2017-01-17:
I use the Racerstar 4in1 20A V1 as well. without mods. It can handle the protocols ok.

It's just the update rate it can't cope with 4K is fine any higher and I get glitches with the motor. Sounds like a tick when it is hovering. 

Thing is I thought I had updated the wiki but can find no history of it, must be going mad.

Eganwp, 2017-01-17:
That clicking noise is being caused by the digital signal being clipped by the capacitor on the ESC. It should always be removed in order to run DShot reliably (even for DShot 150). This same question has been posted by many people both here and on the forums and removing the signal cap always fixes it for people. Please see the wiki links of the oscilloscope pics of why this is happening if you're still curious.

atomiclama, 2017-01-17:
It's on my list of things to try but not that high up as it flys great as it is.

Thing is, the RC filter should effect the protocol but that is not a problem when running at 4k update rate. 
same protocol speed but at 8K update and there is a problem. That does not make sense to me.

closed 2018-04-21 by DieHertz