betaflight/betaflight #2091: SPRACING F3 not able to take off - BF 3.1

Noctaro, 2017-01-11:
Hi, i tested flashing BF 3.1 RC5 and RC7 onto a SPRACINGF3 board. When i arm, it will spin motors at minthrottle but not react to any input. In blackbox i can see the arming beep and my stick movement, but nothing else happens. (rccommand.THROTTLE goes up and down)
The quadcopter model showing motor rpm stays gray. (failsavePhase = 0)
Motor[1-4] stay at min throttle.

I may upload the log if needed. Back to 3.01 everything works fine. Same setup. My Naze works without any problem on 3.1 RC 5.
`
# diff

# version
# Betaflight / SPRACINGF3 3.1.0 Jan  8 2017 / 01:55:49 (e77284c)

# name
name NocRodeo

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# map

# serial

# led

# color

# mode_color

# aux
aux 0 27 1 1300 1700
aux 1 1 1 1700 2100
aux 2 12 2 1300 2100

# adjrange

# rxrange

# rxfail

# master
set min_throttle = 1100
set max_throttle = 1900
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 34
set align_board_yaw = 180
set pid_process_denom = 4
set blackbox_rate_denom = 2

# profile
profile 0

set d_pitch = 27
set d_roll = 25
rateprofile 0

# rateprofile
rateprofile 0

set rc_expo = 15
set rc_yaw_expo = 10
set tpa_rate = 15`

Eganwp, 2017-01-11:
RC5 is working fine on my SPRacing F3 board.

Noctaro, 2017-01-11:
Hmmm.. strange. Are you also using PPM input?

Eganwp, 2017-01-11:
Yes, FrSky SBUS. Try a full chip erase and double check your RX channel
order tab. Haven't tested RC7 but will soon.

Noctaro, 2017-01-11:
I am on Devo 7, as said, on 3.01 i use the same settings and everything works. I double checked rc inputs. Works as intended. Here is the BB LOG -> http://s000.tinyupload.com/index.php?file_id=55563891182329716930

RipperDrone, 2017-01-11:
Maybe it's a feature, not a bug: In BF 3.1, you might have the pid_at_min_throttle CLI switch enabled by default. This will cause all stick inputs being ignored after arming except throttle itself (to protect you from I term build-ups and annoying launch effects, bounces). Once throttle surpasses the airmode_activate_throttle value (e.g. 1350), you will regain all controls.

mhunley123, 2017-01-11:
I have the F3 board and had the exact same problem and it ended up being the Gyro rate and PID loop.
I just changed the pid_process_denom to 5 and it started working.
The only setup that worked for me was:
Gyro update at 2.67KHz
PID loop at .53KHz

Eganwp, 2017-01-11:
Hmm, that's odd. My SPRacing F3 v1.1 is arming and working perfectly on RC5 (haven't tried 7) @ 4/4/16 & 4/2/8. 

4/4 has a touch too much dithering on cycle times for my likings however (due to CPU loading) so 4/2 works best. Can you post a diff for us?

RipperDrone, 2017-01-12:
Baro ON? Accel on? Tried switching them off? What's your CPU load shown in BFC?

RipperDrone, 2017-01-12:
RF rate is >0? There was a BFC Version disconnect so after flashing to 3.1, loading data from older BFC versions set the FC rate to 0. (@Eganwp: your problem seems different, though)

Noctaro, 2017-01-12:
@RipperDrone 
I for sure have exeeded the minthrottle, moved throttle stick up to max. I think you can see it in the bb log.  
CPU hovering around 6%, ACC is on. (Will try to deactivate it)
What does BFC Version disconnect mean? Went anything wrong when flashing the new version? How may i handle this? I flashed it more then 3 times with full chip erase option.

@mhunley123 
Thanks for the hint, will try to give those settings a try.

RipperDrone, 2017-01-12:
@Noctaro I mistyped RC rate - just a multiplier acting on RC stick inputs. By disconnect I meant non-compatibilities btw old version BFC and BF Firmware 3.1, so loading dumps resultet in RC rate unintentionally reset to zero (all inputs active but not having any effect on the quad). 

However, since you can fly it with different looptime / gyro rates, the bug seems to be related to CPU load, RC rate can be ruled out as a root cause.

Noctaro, 2017-01-13:
@RipperDrone 
I see, thank you for your thoughts! If you want me to do further testing, just let me know :)

Cheredanine, 2017-01-14:
Be tempted to check the gyro, I think the origional sprf3 had a 6500/9250, they are now listed as 6050

Noctaro, 2017-01-14:
@Cheredanine 
I checked gyro in CLI Tab, it says MPU6050 on both versions of betaflight. Version 3.01 didnt seem to have that problem. I think it only occurs in BF3.1 (did not test any revision < RC5)

RipperDrone, 2017-01-14:
Could it be motors protocol: DShot, OneShot, MultiShot, ...?

Noctaro, 2017-01-14:
@RipperDrone I tested it with PWM@400HZ and with OneShot. Same behaviour, so i dont think this might be the problem.

Noctaro, 2017-01-17:
Flashed RC9 today, seems standart PID loop time settings changed from 2k/0.5k to 2k/1k, but the problem of ignored stick input persits with these settings.

RipperDrone, 2017-01-17:
get airmode_activate_throttle? Sollte nicht zu hoch liegen...

RipperDrone, 2017-01-17:
ist denn throttle auch tot nach Arming?

Noctaro, 2017-01-18:
Yes, i each time apply more and more until full throttle. No reaction.

Rain, 2017-01-18:
This issue seems to exist on the Lux v2 as well with 3.1 RC9. The motors will only rise above `min_throttle` if either `2.76KHz`/`0.53KHz` or `1KHz`/`0.25KHz` is selected for the Gyro update rate & PID loop time - exactly the same symptoms that @Noctaro has reported.

If anyone needs any additional information, I'd be happy to provide more as well! I figured opening up a separate issue for the Lux v2 wasn't a good idea as I'm assuming this is the same underlying issue; doesn't seem to be SPRACING F3 specific!

Cheredanine, 2017-01-18:
@Rain 
Esc protocol?
Escs?

Rain, 2017-01-19:
@Cheredanine 
ESCs: Lumenier 30A BLHeli_S, flashed to latest BLHeli_S firmware (16.6, I believe)
ESC Protocol: Doesn't matter. Both DSHOT and Oneshot produce the same effect. The motors spin up just fine in the motors tab (and/or through BLHeli Suite).

Noctaro, 2017-01-19:
@Rain Ah ok, thanks for the info, as far as i know your Lux board is also F3 generation and uses other sensors then my SPRACINGF3. So it seems the problem does not depend to the model of gyro.

aowi7280, 2017-01-19:
This is the issue on the RTFQ Flip32 F3OSD board also (BF 3.1.0 unreleased). Gyro update at 2.67KHz
 PID loop at .53KHz works.

Noctaro, 2017-01-19:
@aowi7280 Thanks for your feedback. I changed the name of the issue to F3 flightcontrol as it seems to be a problem with this architecture.

borisbstyle, 2017-01-19:
When you use stick arming your yaw input is ignored on zero throttle. Otherwise you would not be able to arm.
Roll and pitch should work if pid_at_min_throttle setting is enaboled

Noctaro, 2017-01-19:
Ah, i see. However, i put all sticks to all directions without touching pid_at_min_throttle. I am not sure if its enabled by default with the new revisions.  But as mentioned i wonder why gyro data would be 0 on all axis while PID has some data (maybe coming from ACC?). I will give it a try to enable or disable pid_at_min_throttle and report back.

Noctaro, 2017-01-19:
Ok so pid_at_min_throttle = ON is the standart settings of RC9
I tested it right now when set to OFF. And now a intresting thing is happening -> If i raise throttle, the motors will all reduce their speed and spin with less rpm. (You can hear it) It does nothing else, even on full stick deflection on all axis. Feels like minthrottle - 20 or so. They stay at the same value along the whole throttle stick range. Only if i reach minthrottle the motors start to spin up a bit faster. Will do a bblog to see whats going on.

Rain, 2017-01-19:
@borisbstyle The issue doesn't seem to be ignored input at zero throttle. When we're throttling up (which can be confirmed on the receiver tab) after arming, nothing happens. The motors will not spin above `min_throttle` while the throttle is most certainty above `min_check`. I've tested with both stick arming arming with an aux channel, zero throttle response either way (and no roll/pitch/yaw response even if the throttle is 100%).

When I get home this evening I'll flash RC10 and reset everything again, just in case that makes a difference, and document exactly what's going on.

borisbstyle, 2017-01-19:
Yeah try again and please also try it on default settings and rates first

borisbstyle, 2017-01-19:
Also post a new log

Post a short log with
set debug_mode = anglerate

Noctaro, 2017-01-19:
Hey, good news, i flashed rc10 and with standart loopsettings 2k/1k it works again :)

Rain, 2017-01-20:
This definitely still seems to be a problem for me. @Noctaro please try different gyro update rates & loop time settings on your SPRACINGF3 and see if it continues to work as expected.

---

@borisbstyle Here is exactly how I reproduced this on the Lux v2 with RC10.

1. Flashed RC10, full chip erase.
2. Set UART4 to Serial RX, ensured Serial/SBUS was selected, switched to DSHOT150, selected 2KHz for Gyro Update Rate (and therefore 0.5KHz for Loop Time, which is automatically selected), then saved.
3. Set proper channel map (Taranis Radio)
4. Created an ARM mode with AUX1 (a switch on my radio)
5. `set debug_mode = anglerate`, then saved.

Next, I powered everything up, armed, and the motors spin, but do not throttle up. Log, as requested: [LOG00007.txt](https://github.com/betaflight/betaflight/files/718164/LOG00007.txt)

Next, I selected 8KHz (the default after the reflash) for the gyro update rate. 2KHz is automatically selected as the loop time. Saved, tried it again... it works as it should. Log: [LOG00008.txt](https://github.com/betaflight/betaflight/files/718167/LOG00008.txt)

While the above logs are with DSHOT enabled, I also re-tested with Oneshot on RC10; same effect. It seems like select pairs of Gyro Update Rate / Loop Time work correctly, others get this strange "stuck at `min_throttle`" effect.

Also, here is an interesting log (from RC10, obtained during the above testing), but I had this happen on RC9 last night as well. Right after arming, the motors sometimes go **nuts**. *Very* fast oscillations from full throttle to zero throttle. It sounds crazy when it happens! This wouldn't be good w/ props on themotors, I'd imagine: [LOG00005.txt](https://github.com/betaflight/betaflight/files/718168/LOG00005.txt)

Eganwp, 2017-01-20:
The symptoms you describe sound like you're not re-setting "level" on the gyro after flashing...

Rain, 2017-01-20:
@Eganwp The accelerometer is what determines if the craft is level, not the gyro, right? Changing the gyro update rate and/or the loop time (without level calibrating) shouldn't affect it.

The problem is not that the FC will not arm. The board is clearly arming, the arm LED lights up. The motors spin up, but they ONLY spin at `min_throttle` (or the DSHOT idle percentage, if DSHOT is enabled).

Noctaro, 2017-01-20:
@Rain Hey i tested 1k/1k, 2k/1k, 4k/2k, 2.67k/1.33k and they all seem to work (with PWM and ONESHOT)
BUT i tested 1k/0.5k, 2k/0.5k and the problem occurs again. So seems the problem persists with some "looptime/gyro update" settings as you mentioned.
I also tried it without calibrating acc, this does not affect it. I dont have DSHOT compatible ESC for now so i cant do any testing with DSHOT.

Noctaro, 2017-01-28:
Hi all. I tested the new 3.1 stable release with some settings: (on the spracing f3)
working looptimes:
4k/2k
2k/2k
2k/1k
1k/1k
1,14k/1,14k
1,33k/1.33k
1,33k/0,67k
1,6k/0,53k

not working:
1k/0,5k
1,33k/0,44k
1,33k/0,33k
1,6k/0,4k
2k/0,5k
4k/0,5k

For now i have tested a cc3d, a naze and a spracingf3 board . Sadly each of these fc still has problems in the new stable release.

mikeller, 2017-01-28:
@Noctaro: Just to clarify, the 'not working' means no reaction to stick input when using PPM?

Noctaro, 2017-01-28:
@mikeller yes, right. the ones not working do not react to stick input. I also have testflown 2k1k on my spracingf3 today without obvius problems.

Rain, 2017-01-29:
I haven't had a chance to test the stable 3.1 release, but what @Noctaro listed for the SPRACINGF3 matches my results with the new Lux v2 on the last 3.1 RC.

@mikeller I doubt PPM is related to the issue; I'm observing the same problem on SBUS (X4R-SB).

sjhstone, 2017-07-19:
Same problem here using parallel PWM receiver. Will arm when gyro sampling rate and PID loop rate are set at 2.67k/1.33k.

jflyper, 2017-07-19:
@sjhstone Can you be more specific about your setup? It has been more than a half year since this issue was opened, and seems like the issue was fixed in patch releases since the thread has been quiet. Your problem may be different.

Noctaro, 2017-08-05:
Sorry for not updating, the problem was resolved. Should i close this issue?

weltermarcel, 2017-08-22:
I am having the same issue with a PIKO BLX flight controller running Betaflight 3.1.7 (3.2 is not available for my FC). I can enable the motors, they spin at the idle RPM and won't accept any input. I tested the gyro and PID rates mentioned above but had no success.

The trick for me was to enter the PID settings tab and click on save (without even making changes). It'd fly fine until I disconnected the battery and connected a new one. I had to do this every time and I wasn't able to remove the battery after clicking on save.

I just found out that lowering my super rate settings for roll, pitch and yaw from 0.7 to 0.69 solved the problem and now it works every time. I set them to 0.7 again and it is still working...?

I hope this helps someone as a workaround or to solve the issue for the next version.

mikeller, 2017-08-23:
@weltermarcel: Your description sounds like you've got a corrupted configuration. Try backing it up (`diff all`), then resetting to defaults, and restoring it.

weltermarcel, 2017-08-23:
@mikeller before all the above I tried flashing BF 3.1.6 and wrote my older settings (obtained from a dump) and it behaved the same way. It may have to do with a corrupted parameter from my dump file and it maybe it got corrected by setting the PID values to 0.69 and back to 0.7?

I currently don't have time to do more testing but I'll come back with an update.

mikeller, 2017-08-25:
@weltermarcel: The way the settings work is that it is possible to dump corrupted values, but when you are trying to restore them the will fail validation and be discarded. So, to get to a known good configuraton, store a `diff` of your settings, do a `defaults`, then restore your settings.