betaflight/betaflight #2092: Micro minimosd kombini fc vbat

jeffwhu, 2017-01-11:
Hi 

betaflight 1.8.7 pikoblx 3.0.1 microminimosd, had this set up for a while working fine, msp enabled uart1 and battery voltage is flashing 0.0v, minimosd gui set to voltage from fc, enabled telemetry in bf , all set right, connections are all ok, what has changed to cause this tried 2 micro minim osd 
Anybody shine some light on this

Regards

Jeff

mikeller, 2017-01-11:
Hi @jeffwhu.

When you are connecting to the flight controller with Betaflight configurator while a battery is plugged in, do you see the battery voltage displayed on the 'Setup' tab?

jeffwhu, 2017-01-11:
Hi
I didn't notice it, under the vbat button on set up the bottom box is greyed out, so I put another set up on it and the same box read 1.4

mikeller, 2017-01-11:
This was the 'Battery Voltage' indicator with a battery connected?

[image]

If so, the voltage indicator on your board seems to be defective or grossly miscalibrated. Read this and try to recalibrate it:

https://github.com/cleanflight/cleanflight/blob/master/docs/Battery.md

jeffwhu, 2017-01-12:
Hi 
Will do I'll look in the morning , I appreciate your time 

Jeff

jeffwhu, 2017-01-12:
[image]
No voltage with battery plugged in, but showing current draw

jeffwhu, 2017-01-12:
Also getting osd overlay but cannot get into osd menu with taranis, receiver change? Its like this with 2 microminims which would point to another issue

mikeller, 2017-01-12:
Ok, what output do you get from a CLI `diff`?

jeffwhu, 2017-01-14:
Sorry, don't know what a "cli diff" is , how do I find it CLI dump? Then what?
I'm out of my depth about coding
Thanks for your help
Jeff

mikeller, 2017-01-14:
Instead of `dump`, just enter `diff`, and copy / paste the output (instead of printing all the settings like `dump`, `diff` only prints the ones that are not default settings, making it a bit easier to spot if something is not as it should be.

jeffwhu, 2017-01-14:
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# BetaFlight/PIKOBLX 3.0.1 Oct 18 2016 / 10:23:16 (48b7b4f)

# name

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature TELEMETRY

# beeper

# map

# serial
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 2 1 925 1300
aux 1 12 0 1300 2100

# adjrange

# rxrange

# rxfail

# master
set auto_disarm_delay = 20
set serialrx_provider = SBUS

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 
like this??

mikeller, 2017-01-14:
Yes, thanks. Looks like everything with your config is ok, so it's probably a problem with your voltage sensor input  circuit or the wiring.

jeffwhu, 2017-01-15:
Battery solders straight to kombini board, also I have just noticed, I can arm the FCC and have motors running fine, but when I try to get into micro minim odd gui via my radio to alter odd settings via goggles I can't get into it throttle stick up half way and  right rudder and elevator stick pushed fully up, I can't get the odd  scenes to display on goggles.
So it would seem the kombini board is out putting anything at all, is there a clue command to change?.
I have odd overlay because it's showing flight time on screen which proves the odd is working, I've tried two micro minim odd and both do the same thing, my thoughts are, the cv is faulty or a setting needs to be changed.
I'm going to solder a + and - straight from battery to mice minim and see what happens minim has two batterry connector one for voltage monitoring and the second I forgot exactly what that does,, not ideal but should work, I really wanted to do is tune quad through minim odd so I don't have to landing connecting to pc etc ev

mikeller, 2017-01-15:
Ok, then it's most likely your board that is defective. Running Vbat wires to your OSD board for battery voltage measuring is a good workaround for this.

jeffwhu, 2017-01-15:
just think of changing to uart 2 to try and get batt voltage, suppose its worth a go?

mikeller, 2017-01-15:
No. If you don't get battery voltage displayed in configurator, this means that the flight controller is not able to read it, and won't be able to send it to OSD.

jeffwhu, 2017-01-15:
oh right i see, i will close the issue, thanks very much for your time
Ergards
Jeff

closed 2017-01-15 by jeffwhu