betaflight/betaflight #2093: Feature Request: Test screen Frame on ARM (OSD Blackbox)

macht-nix, 2017-01-11:
Hi Guys,

my idea is to let the OSD send an one frame test screen on the begging of the ARM beep so it is possible to sync the blackbox log to the DVR footage without audio or filming the sticks

[image]

jflyper, 2017-01-12:
Displaying asterisk (*) for few seconds be enough for this purpose?

And there will be a maximum skew of 1/60s + alpha between the start of the logging and the mark is displayed. Is this okay?

macht-nix, 2017-01-12:
I think there sould be no link to the log start
it should just be linked to the Arming beep because BetaBlackboxExporer is geared to the Arming beep begining (as seen in the Screenshot)
...so it can be triggert by the same event as the beeper on every arming same as like the beeper is (if its that simpe to code) ;-P

borisbstyle, 2017-01-12:
Just arm indicatio is enough ;)

Osd already can show arm / disarm status

macht-nix, 2017-01-12:
1. is the timing of the arm status right? (sry that i cant test by now i will)
2. if one would like to have clean video with stick overlay arm status is not the best option (i know osd is switchable)
3. i know code lines are limited and for me personaly if the timing of the arm display is right that solution would fit

borisbstyle, 2017-01-12:
Well arming timing is right as far as I know. Once osd shows that its armed the blackbox also started.
I personally dont see a need for adding another additional thing for that.

@macht-nix did you try syncing with arming notification?

borisbstyle, 2017-01-12:
But as far as it goes for blackbox and osd something could be added to show if blackbox is running for example.

macht-nix, 2017-01-12:
> @macht-nix did you try syncing with arming notification?
i cant by now  but hope ill find time next week (if someone was faster go ahead)

mikeller, 2017-01-12:
> if one would like to have clean video with stick overlay arm status is not the best option, a 5 star solution would be the display of the craft name for x seconds (i know osd is switchable)

I think there is a DIY solution for this:
- set up OSD switch on an RC channel;
- on your TX, set this RC channel to a momentary switch of a second or so, activated by the input that controls the arming RC channel;
- profit :-).

macht-nix, 2017-01-12:
@borisbstyle  ...i managed to test it today....

im sorry i was maybe a little bit fast on this idea i sould have tested all before i came out withe it ;-)
everything workes perfekt for me as it is.

i simply disable all elements of the osd and it shows "Armed" in exakt time and length match to the beep 
and after this armig display i disable the osd (as @mikeller sugested) so is does not display the stats when disarming

done

i d say thank you all the issue can be closed

closed 2017-01-12 by macht-nix