betaflight/betaflight #2094: TBS SmartAudio on OMNIBUSF4SD not working

DerOrakel, 2017-01-11:
Hi,

I wanted to test out TBS SmartAudio so i got a TBS UNIFY PRO 5G8 HV - RACE. I'm using an Omnibus F4 V2  FC with the OMNIBUSF4SD target. Reciever is a Flysky X6B (using IBUS) connected to UART1. I tried using UART6 and UART3 for TBS SmartAudio (audio cable connected to the TX pin) and configured the port in the GUI (1.8.7) using BF 3.1 RC6 and RC7. I can not get Smart Audio to work on either port. In the VTX CMS page there are just dashes where i believe should be Band and Channel shown, i can open the STAT item and see that there is no response vom the Unfiy just packets being send.
I have read the TBS SmartAudio Pull Request and Wiki page and there was some discussion about the RE1 that is a F4 board also but after looking at the source to me it seems these changes where already merged.

jflyper, 2017-01-12:
@DerOrakel 
I just confirmed it working on OMNIBUS F4 V2 + Unify 5G8 Pro and Unify 5G8 Pro HV with RC7.

Did you confirm the TX pin position?

DerOrakel, 2017-01-12:
Hmm had that double checked yesterday but I will confirm it again when I
get home. Also maybe it has to do with that the race does SmartAudio v2 vs
v1 on the hv?

jflyper, 2017-01-12:
@DerOrakel 
Oh, you got the race edition... TBH, the code has never been tested with the race edition; I was told that it speaks V2 protocol, and driver can't distinguish race and non-race editions.
But the race edition is apparently a variable...

DerOrakel, 2017-01-12:
Ok I just checked again for the pin and I'm 100% sure that I'm using the
correct pin it is tx3 (the one in the same line as crnt and gnd, if look at
the board from above with the usb facing towards you is the most left row,
fourth pin from the top) so the problem seems to be with SmartAudio v2 in
my opinion. TBS SmartAudio is enabled for uart3 Baudrate is set to auto,
stat menu shows something like 4900baud.

jflyper, 2017-01-13:
@DerOrakel I just confirmed it working, again, with UART3.

So, it's either something different with the race edition, or your board and/or configuration...

lichtl, 2017-01-13:
Hi,

i yesterday received some RACE vtx. Currently using a HV on my OMNIBUSF3 so I can test on the weekend if RACE/ non-RACE work on F3

jflyper, 2017-01-13:
@lichtl Cool! I'm looking forward to hearing the results!

jflyper, 2017-01-13:
It may be a baud rate problem.
#2108 is an experimental PR so users can tweak the baud rate scan range.

lichtl, 2017-01-13:
@jflyper just tested on the OMNIBUS with a out of the box TBS Unify Race. Was working without any issues.  Channel switching / TX power all was working. Only noticed that i was able to change to 500mw/800mw but on the status line it stayed at 200.

Interesting was also that changes through the OSD unlocked the Unify channel+tx power

jflyper, 2017-01-14:
@lichtl Thanks for the report.
Yes, it is unlocked by default, so it is solely up to the user to be compliant with respective local regulations :)

The behavior on power is as expected.

@DerOrakel 
Someone found that there's some resource collision on UART6 with OMNIBUSF4. Hang on.

DerOrakel, 2017-01-14:
@jflyper it's propably the hardware inverter used for the sbus that's why
I'm using uart3.
My guess still is with this being an issue similar to that with the
Brainfpv RE1. I will try to test your pr but I can not do that before
sunday any suggestions for baudrate and or scan rate?

jflyper, 2017-01-14:
@DerOrakel Inverter is on UART1, and it only works for signals from SBUS/PPM pin.

As for RE1, I think someone already have a working code. It was missing BIDIR_PP mode support.
I don't remember issue number thought...

jflyper, 2017-01-14:
- Above mentioned case with HV + OMNIBUS F4 turned out that it is most likely the problem with UART6 hardware itself. It doesn't connect to MSP via FTDI.

stellarhopper, 2017-01-14:
Just chiming in - I know someone who is also running smartaudio without problems on the omnibusf4 v2 pro + unify race combination.

For brainfpv RE1, the latest 3.1 RC builds have all the right support for it, and the RC builds should work fine - I've been running it for a while now.

RC-Slater, 2017-01-22:
I'm running omnibus f4 v2 pro with the unify pro race without any issues.  Here's my video demo. https://youtu.be/Klz56bvTkZc

DerOrakel, 2017-01-22:
How is your unify pro connected? Power, gnd, video and so on I mean.

>

RC-Slater, 2017-01-22:
Power ground and video to the power ground and video header on omnibus (12v feed from pdb)  The SA wire is running to tx pin on uart6.

RC-Slater, 2017-01-22:
Then just enabled smartaudio in configurator on uart6

DerOrakel, 2017-02-14:
Update: never got it working with the OMNIBUS F4 V2 but that may have been a faulty board anyway. It died a week later. I now have a OMNIBUS F4 Pro and Smartaudio works fine using 3.1.5

closed 2017-02-14 by DerOrakel

vastFPV, 2017-04-23:
sorry if you answered this already, but you are using the non-inverted s.port signal from your receiver? you cant just hook up s.port regularly like that on the F4. there is only one hardware inverter and it is used for Sbus. so there is a hack that you solder directly to the non-inverted signal