betaflight/betaflight #2095: arming issue with atom v2 and betaflight

buzznlush, 2017-01-12:


buzznlush, 2017-01-12:
i have a rotorx atom v2 i cannot get to arm unless i first plug into the pc and save and reboot in beta flight, i can then arm and alls good, when i disconnect the battery from the quad and plug back in it wont arm again until i go through the same above procedure .
i have flashed betaflight 3.0.0 atom v2 to the fc, i have light control from the tx all the time but arming will only take place doing what was before explained

mikeller, 2017-01-12:
Can you try Betaflight 3.1.0 please?

buzznlush, 2017-01-12:
# version
# BetaFlight/COLIBRI_RACE 3.0.0 Oct  6 2016 / 09:00:01 (f408ed9)

# name
name -

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature VBAT
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature FAILSAFE
feature LED_STRIP
feature OSD
feature AIRMODE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
[... 349 lines cut ...]
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 70
set rc_rate_yaw = 80
set rc_expo = 15
set rc_yaw_expo = 15
set thr_mid = 50
set thr_expo = 0
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set tpa_rate = 20
set tpa_breakpoint = 1650

# status
System Uptime: 24 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:26%
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
Cycle Time: 135, I2C Errors: 1, config size: 1476

#

buzznlush, 2017-01-12:
i have no idea what im dong on this site i am sorry

buzznlush, 2017-01-12:
Hi Chris 
i have been using tbs agent for uploading, i had a look but it wasnt showing unstable versions

mikeller, 2017-01-12:
Then use the Betaflight configurator.

buzznlush, 2017-01-12:
okay. i will look into that. i did not expect such an issue from a arf quad, build bind and fly was supposed to be it, ive been sitting at a table for a week and it hasnt left the table

buzznlush, 2017-01-12:
i have found it, i could load the firmware but i was getting a message saying failed to open serial port , i wasnt sure weather i had to hold the bootloader button in while plugging usb

mikeller, 2017-01-12:
There is two ways to do it:
1. don't hold the boot button and don't enable 'No reboot sequence'.
2. hold the boot button, and enable 'No reboot sequcence'.

1. is easier, 2. is a last resort if 1 doesn't work.

buzznlush, 2017-01-12:
building is one thing software is another,,, thats why im here :-)

buzznlush, 2017-01-12:
will i have lost settings such as pid settings etc, or they come set with new versions of software

buzznlush, 2017-01-12:
i have tried scenario 1 i get a repeated failed to open serial port. i will try option 2

buzznlush, 2017-01-12:
same deal, perhaps im doing something wrong, i cant see what but im getting the same error message

mikeller, 2017-01-12:
https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

buzznlush, 2017-01-12:
awesome thank you muchly, a little later his arvo i will look at following instructions.
do you feel the arming issue is related to the atom v2  v3.0.0 software or feel its a better version than the last. big learning curve for me but im determined

buzznlush, 2017-01-12:
i have not been through the osd set up as i do not own goggles and am still flying los, would this be contributing to my issue do you think ?

RipperDrone, 2017-01-12:
Definitely. On the TBS hardware on the Atom, you need to go through the OSD setup once before it will arm. It's a TBS Powercube specialty, not related to BF.

buzznlush, 2017-01-12:
thanks forthwt info I just tried it by pressing the button on he cube and
calibrating manually. I didn't think it worked but my min throttle was too
high. Now it's all peachy.
Thanks everyone for your offers of help.

Muchly appreciated

Darren

RipperDrone, 2017-01-12:
Glad it worked 4 u. Pls close this issue when u have  time.

buzznlush, 2017-01-12:
I'll be on to it shortly ☺

Darren

closed 2017-01-12 by buzznlush