betaflight/betaflight #2107: VTX integration/selection scheme

jflyper, 2017-01-13:
In addition to TBS Unify SmartAudio, we are about to integrate another off-board intelligent vtx, IRC Tramp, and we should start discussing how we should handle them in BF. We also have two other FC controllable vtx that are somewhat target specific (options VTX and USE_RTC6705), but our immediate and short term (for 3.1 Release) goal is to decide how we handle SmartAudio and Tramp together.

- Build option
USE_SMARTAUDIO is currently enabled for all >128K targets, except for those having either VTX or USE_RTC6705. Shall we do the same for USE_TRAMP?

- Configuration
TASK_VTXCTRL task is always active for those with USE_SMARTAUDIO, and SmartAudio process do it's job only when one of the port has FUNCTION_VTX_CONTROL mask (i.e. 2048) and has been initialized appropriately.

The simplest way to accomplish our goal is to
1. Include USE_TRAMP for all >128K targets.
2. Assign another serial function mask for Tramp (e.g. 8192)
3. Admit small overhead in TASK_VTXCTRL from calling an unused handler of the two.

Any thoughts?

raphaelcoeffic, 2017-01-14:
I'm ok with this. Anything else would probably introduce more work and need some bigger master plan.

But how do we handle that in CMS? Does that mean we would have multiple VTX menus?

jflyper, 2017-01-14:
> But how do we handle that in CMS? Does that mean we would have multiple VTX menus?

Yeap. VTX SA and VTX TR :)

raphaelcoeffic, 2017-01-14:
Ok, I could imagine living with it for a while, but in the long run, it would be have to be consolidated, I think.

Be the way, same issue with MSP once again, right? Or do we want to duplicate these functions as well? I'd rather not. And still, that would require a mechanism for determining which of the supported VTXs is really installed and running.

raphaelcoeffic, 2017-01-14:
@jflyper couldn't we init CMS and MSP based on the serial config? VTXs using direct RT6705 are then only initialized when no serial config for another VTX is present. I guess that could work, right?

raphaelcoeffic, 2017-01-14:
@martinbudden, @borisbstyle any preferences from your side?

raphaelcoeffic, 2017-01-15:
@jflyper I've already implemented it this way #2106. Hope it's OK this way. I guess it will "beautified" in #2114.

raphaelcoeffic, 2017-01-18:
@jflyper I guess we close this one, right?

jflyper, 2017-01-26:
Yeap.

closed 2017-01-26 by jflyper