betaflight/betaflight #2111: Spektrum quad race serial receiver always on bind mode when turned on with F4 Omnibus OSD

CraigFPV42, 2017-01-14:
Hi,

I have a spektrum 4649T (http://www.spektrumrc.com/Products/Default.aspx?ProdId=SPM4649T) quad race serial receiver and an F4 omnibus with an integrated OSD hooked into UART 1. When power is applied the receiver goes straght into bind mode. I can bind it and fly it without an issue, just every time I unplug the battery and plug in a new one I will have to bind it again. 

Is there anything in the CLI I can do, or is this an issue with the .hex file?

AndersHoglund, 2017-01-15:
How is the Satellite connected and powered? This Rx requires +5V minimum. See also #2120

CraigFPV42, 2017-01-16:
It's into UART 1, it gets 5V.

AndersHoglund, 2017-01-16:
OK, great. The original OP Revo design has a 10K pulldown on the UART1 Rx input. If the OBF4 is a true clone that might be case for you too. And 10k might be enough to force the Sat into bind. 
Move to another UART, and/or connect to the UARTx Tx line for FC telemetry. Select SRXL in that case.

CraigFPV42, 2017-01-16:
So right now I have it plugged into the UART 1 RX. 

What you're telling me is to try UART1 TX, or change to another another UART all together? I will give it a try today and report back.

CraigFPV42, 2017-01-16:
Also interestingly enough if I flash the AirbotF4 hex file onto it the OSD stops working but the bind issue goes away!

jumatic, 2017-01-23:
Same problem here. F4 Omnibus Pro V2. It gets automatically into bind. Also using it on Serial RX1 J7 will cause chaotic rc output. On the Flip32 F3 with spf3 FW it works like a charm. Can use same port as SBUS but connecting signal line to tx. 
Tried TX1 from J5 Port but its not assigned as resource. Using RX1 from J5 works but with autobind. Also telemetry functions not tested.

AndersHoglund, 2017-01-23:
If you are using SRXL, you must connect to the Tx pin of the UART.  Any free UART.

As TX3 on J10 seems to work for you, use it.  Also Tx6 on J10 should work. 
J5 Tx1 should work for SRXL, but you say "no resources". I do not really understand that.  Are J5, J7 and J8 all UART1? Seems so... 

As J7 are for both SBUS and PPM I think there are some solder bridges to remove to select either PPM or SBUS(SerialRx), you can NOT have both. Then there will chaos. Might even be the reason for constant re-bind. I think it was R19/R20 on the original AirbotF4 that select PPM or SBUS on J7. If J5, J7 and J8 also are interconnected, those solder bridges will affect all three.

It is hard to understand all the conflicting and incomplete info on rtfq site. It is a mess, really. You have to figure out how things are connected, using a DVM or ask them.  I do not have any Omnibus.

closed 2017-03-15 by AndersHoglund