betaflight/betaflight #2112: Start-up problems Omnibus F4 and ESC Calibration after power up

CrazyKoboldFPV, 2017-01-14:
Good Morning!
I have a problem that is not solvable for me.

If I want to disarm the quad, the ESC starts with a calibration. The video you can see here: https://www.instagram.com/p/BPFvpCdjhMD/
The problem occurred after I connected LEDs to PWM5 (signal) and + & - on the FC to LED + & -. Configure on Aux5 for color change via the FrSky Taranis. It is working. Unfortunately, I had additionally changed the RC rate and super rate.

As soon as I connect the Lipo, the ESC starts to beep as if they were calibrating. The quad can no longer be disarmed. If he can be disarmed, then only 1-3 motors will run.

If I close the quad on the computer, the Min + Max Throttel has changed and I have to calibrate again. Then it works for 4-5 lipos, without changes it comes then that the whole thing starts again from the beginning.

I have already checked all solder joints and could not detect any errors.

My Hard & Software:
BF: problems started on 3.1 nightly build and are now also available on the last 3.1 OmnibusF4
FC: Omnibus F4
Receiver: FrSky X4RSB with telemetry
LED: LEDs on PWM5 (signal) and + & - on the FC on LED + & - connected

I hope you can help me.

Dump:
# dump

# version
# Betaflight / OMNIBUSF4 3.1.0 Jan 11 2017 / 08:57:55 (322ae59)

# name
name -

# resources
resource BEEPER B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP A01

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
[... 448 lines cut ...]
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_stick_sensitivity = 100
set level_angle_limit = 70
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 115
set rc_rate_yaw = 115
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 10
set tpa_breakpoint = 1650

#

RipperDrone, 2017-01-14:
Just to tprule oht a simp!e one: BLHeli has a tick box 'stick calibration' that can be ticked off, so stick inputs won't send config commands to the ESC anymore...

CrazyKoboldFPV, 2017-01-14:
Thanks for the tip. I've attached a screenshot of how the ESC configuration looks after the problem:

[image]

Afterwards, I just dragged the configuration and set "Programming by tx" = Off. Screenshot:

[image]

Afterwards, I was able to start the quad 7 times and successfully disarm. At the 8 times the buzzer of the FC again gepiepty, the connected LED stayed on and the FC has blinked blue. Disarm did not function.

RipperDrone, 2017-01-14:
Maybe your ESC timers are borderline. Try to set them min ppm throttle to 1040 and max to 1960 in BLHeli, instead of 1000-2000. One or the other of the ESCs might not be able to reach up/down to the thresholds reliably because of timer jitters...

CrazyKoboldFPV, 2017-01-14:
Now the FC behaves as in the video. Sometimes immediately after the lipo insert, sometimes also after several attempts. The beeps come from the buzzer. The esc do not beeep as in the first video.
https://drive.google.com/file/d/0BxavptpwxT_sM3lJeFlmcXVqdVU/view?usp=drivesdk

@RipperDrone  I will try it. But would not the problem then do not arise with each insertion of the Lipo? Would the FC report the problem of esc?

mikeller, 2017-01-14:
> The problem occurred after I connected LEDs to PWM5 (signal) and + & - on the FC to LED + & -. 

Could it be that you connected the LED to 3.3V on the FC? This would probably overload the 3.3V voltage regulator and cause all sorts of problems. Do you have an external PDB with 5V regulator? If so, it's probably best to hook up the LED supply to that.

RipperDrone, 2017-01-14:
@mikeller Good point - on my F4 Airbot setup, I only hooked the FC up to VBat on the F4 board itself, all other consumers (cam, VTX, even RX) are driven off of the PDB. No LEDs like @CrazyKoboldFPV. Regulator seems to run into hot overload when powering more than 100-150mW, depending on cooling situation.

CrazyKoboldFPV, 2017-01-14:
I connected the LED to the LED pada on the FC. The manual shows were the LED could be connect. I had measured it before and it came to 4.7 volts. But I can solder it to the test to the PDB.

RipperDrone, 2017-01-14:
...or simply switch LED support off in BFC for testing...

CrazyKoboldFPV, 2017-01-14:
That would have been too easy. I did not think about it. Have it ready soldered and will now start a test. But the first Lipos seems working.

RipperDrone, 2017-01-14:
Here'S a source where you can read about the maldesign of the Airbot F4's power circuit: You can always hook it up to a 5V source and run a lot of consumers off the FC, but then you will not have the VBat monitoring function available. On the other hand, if you power it by VBat, there is such a high voltage drop's power to be dissipated by the LDO regulator, it will overheat quickly.

https://www.rcgroups.com/forums/showthread.php?2731917-Airbot-F4-Flip32-F4-Thread

CrazyKoboldFPV, 2017-01-14:
I ran the FC to Lipo Power. But because of the heat the onboard LDO by a Pololu exchanged. Then there are no heat problems and you get the vbat info for the Telemetry. I have now run the LED on the PDB. I also soldered an earth cable of an ESC. Currently the error is gone. 

However, it takes sometimes 4-5 seconds until it takes the disarm signal and starts the engines. Does not beeep, but the blue LED flashes. But it runs. So no software problem as suspected.

RipperDrone, 2017-01-14:
Ok - no delays here on my two Airbots - strange... large capacitors somewhere in the power supply circuit?

My power setup:

- Matek PDB powering RX (S-Bus + telemetry pins + GND to FC, but 5V pin from PDB), VTX, FPV camera
- VBat from Matek board (4S LiPo) to supply FC (VBat pin + GND)

Arms/disarms instantly.

mikeller, 2017-01-14:
The delay could be caused by your flight controller running above 100% load, which stops it from arming. Try reducing the PID loop frequency to see if that fixes it.

CrazyKoboldFPV, 2017-01-15:
Gyro and PID loop both run with 4khz. The processor has idle 2% load when telemetry and Led on. Cabling is now FC with vbat, frsky telemetry & LED signal. Camera is connected to the vtx and the vtx gets power from the PDB. Just as now the LEDs get also current from the PDB.Problem is also here again, that it does not appear every time. It also makes no difference how far the remote is away. I had already thought of a problem. But I already see the vfas at the remote before I can disarm.

CrazyKoboldFPV, 2017-01-15:
No capacitor in the circuit.

CrazyKoboldFPV, 2017-01-15:
Could be due to the fact that I did not place the Copter directly on a straight surface, but held in the hand. Does he initially calibrate the gyro and can not be disarmed? I place it directly, it works. Now I can see no mistake. Seems first well in 20 attempts to connect the Lipo. Thanks a lot for the nice and competent help!

closed 2017-01-15 by CrazyKoboldFPV

RipperDrone, 2017-01-15:
Blue LED blinking 10x in your videos is definitely repeated reset/initialization of the FC - there has to be a voltage supply problem in your wiring. Pls measure resistance VBAT to GND - it seems like you have a current drain somewhere. Maybe VBat takes some time to stabilize and allow arming (has to be above VBat_min set in the BF Configurator to arm). Maybe solder splashes? Or bad insulation to GND / frame parts somewhere?

I'm afraid that if you find nothing else you will have to try another FC?

RipperDrone, 2017-01-15:
Oh man, just now I read your explanation: Yes, there is a small_angle defined in the CLI, arming only authorized once the quad is level within that angle threshold (and at rest - no accel values). So you got it resolved 100% now, right?

CrazyKoboldFPV, 2017-01-15:
During my test I could not find any problem. It seems to have been a problem with the power supply of the LED. Thanks again for the help!