betaflight/betaflight #2113: Problem with osd kiwi f4 flight controller

michael-blue2, 2017-01-14:
Recently received my new fc board and install it on, everything works perfectly running multishot 8k/8k/32k . The osd in the configuration is enabled and the tab with the setting of the osd is visible. I can choose whatever data to be shown at the osd menu and until that everything is fine. The problem is that out of my goggles i get a blank black screen. I have connected my vtx with my camera directly to each other and they work like a charm but when i connect them with the board vin and vout are outputting a blank screen even without the OSD data on them, if i disconnect the camera and leave only the vtx nothing change. I have tried betaflight 3.0.1 and 3.1 rc7 but still i cannot solve the problem.
Could this be a firmware problem or a hardware problem? 
What tests should i do to be sure?

Gaz-B, 2017-01-14:
Have you chosen a font if not you must.

jflyper, 2017-01-14:
I suspect CAM output is connected to VOUT and VIN is connected to VTX.

michael-blue2, 2017-01-14:
i have not choose a font but by vtx is transmitting a blackscreen/ or just nothing. Its not like i just dont see the data on my screen, but theres absolutly nothing
about vout and vin, i have connected vin the camera and vout the vtx like pdf shows of kiwi f4 fc

Gaz-B, 2017-01-14:
On mine the screen appeared to be solid with Vs before selecting a font I could just make out the video behind them though. personally I think it sounds like a hardware issue contact John Lemon he is very helpful.
ms@flyinglemon.eu

BitSnake1988, 2017-02-13:
Hi all,

I've got exactly the same issue, i just wrote to flying lemon to see if they can help.

i tried to replace cam and VTX without any improvements.
I even added a condenser on the main power supply to filter the ESCnoise...

I'll let you know if i got any news

jflyper, 2017-02-13:
For those of you having "BLACK/BLANK screen" on KIWI F4, can you upload a hi-res pic of the board containing this area?

[image]

BitSnake1988, 2017-02-13:
[image]

jflyper, 2017-02-13:
Components and soldering look okay...

There were cases with omnibus that users had a blank screen caused by missing decoupling capacitor and/or input load resistor. But your cases is apparently different...

BitSnake1988, 2017-02-13:
I did not wire anything on those pads, i just pre-soldered the data pad by mistake..
i do not use any serial interface.

michelmorin, 2017-03-09:
What font did you guys pick and how do you change it.   I'm a little confused.    I do get the cam image in the background but I get all the Vs on the screen too

Gaz-B, 2017-03-09:
From memory I clicked on bold then upload I think.

michelmorin, 2017-03-10:
Awesome.   Thank you.   Tried various ones and they all worked but picked default as that is the one I prefered

LuctenFinger, 2017-03-20:
I've got exactly the same issue on OMNIBUS F4 PRO , too
i tried to change the font without any improvements.....

floggi, 2017-10-01:
Same issue here with a Holybro Kakute F4. Just receiving black screen on my goggles (with OSD data). If I connect directly (cam + vtx wire), everything fine (of course no Betaflight OSD data display.

jflyper, 2017-10-02:
@floggi  If you are not getting any picture nor overlay, then it is the hardware issue, either broken overlay chip or bad build.

## If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

closed 2017-10-02 by jflyper