betaflight/betaflight #2115: omnibis f4 spektrum issue

insainone, 2017-01-14:
Setup.

Martian ll frame,
Myairbot omnibus f4,
Diatone PDB 5v, 12v bec, lc filter,
Xracer Quadrant 24a escs,
Hs1177,
ReadymadeRC pro vtx,
Cobra 2205 2300kv,
Turnigy graphene 1300,
Betaflight 3.1,
D shot 600
8k, 8k, 
3% cpu load,

The quad flies great as long as no quick inputs are given. When a fast input or full stick deflection is given all reciever transmission stops. The motors spool down and the quad ceases to operate. I have to power cycle the system to regain communication between the radio and aircraft.

jflyper, 2017-01-15:
For isolation, can you test your "fast input / full stick deflection" without powering the esc/motors?

siamect, 2017-01-28:
I got to give a "Me too!" on this one. I had the problem both on 3.0 and  on 3.1. I don't know when it started but I just received a Taranis Q X7 and while playing with it, it happened. The motors were not armed.  
 
It is repeatable and typically it happens when you slam the stick to max or min both of them at the same time. I have limited the output in the Taranis to 1001-1999 or something so it should not go outside the 1000-2000 values. 
I upgraded to 3.1 and still the problem is there.

But there is a difference. My FC reboots and comes up again or at lest the motors make that start-up do-da-di sound.  It has never happened during flight, but I'm a gentle pilot... I happens even when I'm connected to the Configurator.

I use a cc3d revo F4 bangood knock-off. The receiver is X4RSB.
/Martin

nathantsoi, 2017-01-28:
anybody have a blackbox of this happening?

insainone, 2017-01-28:
I figured out my issue. I was having brown out issues at high amp draw. I found that I had powered the board on a 5v output pin instead of an input. This is what caused the issue. It fried the regulator. I explained what happened to myairbot and they very kindly are sending me a free replacement since it was the pinout on there site that didn't show that this was a output. Very nice people and I still believe in omnibus.

closed 2017-01-28 by nathantsoi