betaflight/betaflight #2116: Custom Motor Mix Dshot Motor 6 not working KISS

beyersito, 2017-01-14:
I am having a weird issue with bf 3.1, I have a kiss FC whose pads 1 2 dont work, flashed bf for custom motor mix using pads 5 and 6. I wanted to use dhsot, so i flashed again kiss firmware, updated the escs and I tested dshot on the kiss firmware with hexacopter config, al motors worked fine, went back again to bf, motor 6 doesnt spin on dshot, it does on oneshot. What might be happening?

mikeller, 2017-01-14:
What output do you get for `resource` and `resource list` commands in CLI? Also, your desired configuration can be achieved easier if, instead of using a custom mixer, you use the QuadX mixer, and then remap the resources for motors 1 - 4  to the pins for motors 3 - 6.

justinledwards, 2017-01-16:
On pikoblx with dshot with remap I'm getting motor 3 not working.  works fine on other modes
standard quadx mixer

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: BEEPER 
A01: FREE 
A02: FREE 
A03: FREE 
A04: MOTOR 3
A05: ADC_BATT 
A06: MOTOR 1
A07: FREE 
A08: FREE 
A09: FREE 
A10: FREE 
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: MPU_EXTI 
B00: MOTOR 4
B01: MOTOR 2
B02: FREE 
B03: SERIAL_TX 2
B04: FREE 
B05: LED 2
B06: SERIAL_TX 1
B07: SERIAL_RX 1
B08: FREE 
B09: LED 1
B10: FREE 
B11: SERIAL_RX 3
B12: MPU_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: FREE 
C14: FREE 
C15: FREE 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

Currently active DMA:
--------------------
DMA1 Channel 1: FREE
DMA1 Channel 2: MOTOR 4
DMA1 Channel 3: MOTOR 2
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 1
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

mikeller, 2017-01-16:
@justinledwards: Read https://github.com/betaflight/betaflight/wiki/DSHOT%20ESC%20Protocol and follow the steps in there for the remapping.

borisbstyle, 2017-01-19:
fixed in RC10

closed 2017-01-19 by borisbstyle