betaflight/betaflight #2118: 32k Gyro Sampling Enabled Arming Issue (Airbot F4 target)

xluckynumber7x, 2017-01-14:
I have a Flip32 F4 w/ MPU9250 gyro that I loaded the 3.1 RC7 Airbot F4 hex onto. I enabled the 32k gyro sampling option and set it to 32k / 16k. Saved and rebooted, 34% cpu, 35 loop time, motors work in motor tab, rx tab shows correct inputs, arming set in modes tab. My quad will not arm with these setting. Tried a lower loop time (8k/8k) with the 32k gyro sampling enabled and still no joy. As soon as I disable 32k gyro sampling it will arm no problem at 8k/8k.

mikeller, 2017-01-15:
It's probably the higher gyro noise when at rest triggering `moron_threshold`. Try increasing `moron_threshold`, or updating to RC8, which contains a higher default.

xluckynumber7x, 2017-01-15:
Thank you, raising the moron_threshold to 50 was what it took to arm with 32k sampling enabled. Looks like I still have some more tinkering to do though as the motors sound super rough and heat up in just a few seconds of being armed. The result is the same for any gyro sampling rate / loop time from 32k to 8k with 32k sampling enabled.

RipperDrone, 2017-01-15:
@mikeller: 
This means sampling rate setting of 8k is overridden when check box 32k is enabled?

@xluckynumber7x:
Seems you have some noise issue, so probably spending some time on 
- reducing vibration
- setting the filtering correctly (notch filter threshold, to be determined by blackbox recorder, see WiKi)
will improve your motors temperature.
I'm assuming you have not dialed in an excessive D term in your PIDs... Good luck!

mikeller, 2017-01-15:
@RipperDrone: No. The dropdown is not for the sampling rate, but for the speed that the gyro update loop runs at. If 32kHz sampling is enabled, and a frequency lower than 32kHz is selected for the update loop, the samples received in between the update loop runs are discarded.

closed 2017-02-06 by xluckynumber7x

mpkiteman, 2017-02-22:
I've got exactly the same issue than xluckynumber7x.
Motors at 32K are running very hot.
And question dampening I'm now an expert because I've got several builds with revolt FC ;-)
So my Airbot is mount with special silicon grommets, the same than revolt.
Still trying to do some experiments...

frizzle101101, 2017-06-15:
The same for me but it let me arm and my esc burned because the motors were dsyncing right from low throttle. I am using the new MatekF405 and lb30_s escs so I didnt expect this was wondering if some one could shed light on the issue.

Oriolkun, 2018-03-10:
Same exactly. 
The option to "activate 32k turn sampling". If I only have the option multi ONLY, arm me before I plugged into the USB of the PC (without having to be connected to BetaF necessarily) and then the lipo. Even though it is in 32k16k, 16k, 16k, 8k8k or 4k4k, it does not matter. It only arms me when I first plug in the usb and then the lipo. On the other hand with the option "activate gyro 32k .." removed arms perfectly whether or not connected by USB.
I've tried the moron_threshold at 48, 100 and 200 and the problem persists. 
I have plugged a buzzer and when it fails to arm an specific sound is released. 
I have noticed the initial "beeps" are faster when you have pluged the USB before the lipo.