betaflight/betaflight #2120: Naze32 Rev6 Spektrum Satellite Stuck in Bind Mode

teozfrank, 2017-01-15:
As the description, everytime the quad is powered on the satellite SPM4649T is stuck in bind mode, even though satellite bind mode is set to 0. Satellite will bind fine with a Rev 5 acro board I have, could anyone shine some light on this? I even tried binding with the rev 5 fc and plugged it back into the rev6, same thing power it on, and it keeps flashing like in bind mode.

AndersHoglund, 2017-01-15:
Spektrum bind has never worked with Nase32 Rev6 due to the Sbus inverter making the UART2 Rx line uni-directional. Rx line must be bi-directional for binding to work. But that should not cause the satellite to constantly enter into bind mode. Strange. How have you connected the Sat? You can not use the Sat connector as the supply voltage there is only 3.3V for ordinary satellites, 4649T requires +5V minimum.

Hint: Connect to UART2 Tx line and select "SRXL" protocol and you will get Telemetry!  Vbat is then reported back from FC to Tx and you can skip the voltage sensor on the satellite.  This does however solve any binding issues. When #1766 is completed it might work better, but still not for Rev6 UART2 Rx line due to the HW design.

teozfrank, 2017-01-16:
How have you connected the Sat? You can not use the Sat connector as the supply voltage there is only 3.3V for ordinary satellites, 4649T requires +5V minimum.

Im powering it with 5v. using the 5v out from the pin above pin one. I currently have the signal wire connected to port 4 on the FC. I did have it connected to the signal pin for the satellite dedicated on the board.

Hint: Connect to UART2 Tx line and select "SRXL" protocol and you will get Telemetry! Vbat is then reported back from FC to Tx and you can skip the voltage sensor on the satellite. This does however solve any binding issues. When #1766 is completed it might work better, but still not for Rev6 UART2 Rx line due to the HW design.

So connect the rx signal line to UART2 tx line and select SRXL protocol and it will work or? I need UART2 tx and rx for my osd board or can it be split?

AndersHoglund, 2017-01-17:
There are probably some pulldown resistor on the inverter on UART1 Rx input, same as Revo, OB etc etc. Use UART1 Tx line and select SRXL serialrx_provider via the cli. No pulldown there. The UART2 will still be free for OSD use.

AndersHoglund, 2017-01-17:
Hmm, the Naze32 is really not suiable for what you want it to do. There are only 2 UARTs. You would need 3 for MSP, SerialRx and OSD. Unless you can combine some functions, or use Softserial.
To fix the constant binding problem, you could try to locate the pull down resistor on the UART2 Rx input, and simply remove it.

teozfrank, 2017-01-17:
How would I use softserial?

How would I locate the pulldown resistor? Is the UART2 rx input on the pad for the satellite or where is it located?

AndersHoglund, 2017-01-17:
Follow the trace from Sat connector Rx pin with a DVM buzzer. But this is tricky stuff if you are not comfortable with hacking micro electronics. If you remove something you must be absolutely sure its the right component. And the operation is hard to revert.....
Get another more suitable FC.

teozfrank, 2017-01-18:
I would be comfortable doing this. Removing the pullup resister, what will this fix, the binding issue (not being able to bind) or the binding mode everytime the fc is powered on?

AndersHoglund, 2017-01-18:
That was just an idea for trouble shooting and experimenting.. A pulldown resistor MIGHT cause the sat to go into bind mode, but I do not know that for a fact. But then you mention you share UART2 with OSD, that is also a candidate. Anything pulling the Sat data line low when powering up will force it into bind mode, you just have to figure out what it is. If that OSD is using both Tx and Rx of UART2 I can't see any way to also connect a sat there. If you want telemetry from FC, that Sat also needs the Rx/Tx lines combined into a bidirectional data line. No way you can share that with an OSD.
Get a FC with more UARTs.

teozfrank, 2017-01-18:
At the moment I dont have the osd connected to UART2 and it still going into bind mode. If I were to use UART2 as the rx input how would I do that?

AndersHoglund, 2017-01-18:
It must be on UART2 already. UART1 is connected to USB. Finito, no more UARTs to choose from. And you can  not share serialRx with USB/MSP.
UART2 Rx line has that pulldown resistor. That is why I suggested to connect to the Tx line instead, it does not have any pulldown. When connected to the Tx line, you must select "serialrx_provider = SRXL" for things to work. I think that SRXL is also available in the GUI in BFC 1.8.9, as from today.

cstengel88, 2017-01-31:
But isn´t this receiver shipped with Autobind so it automatically enters bind mode if it can´t connect to the last target it was bound to ? Does this also apply if the remote is powered on first?

closed 2017-03-15 by AndersHoglund

teozfrank, 2017-03-15:
This issue was due to a bad board, I got a new rev6 and it works perfect.