betaflight/betaflight #2123: Flip32 F3 OSD board does not connect after flashing SPRACINGF3

aowi7280, 2017-01-15:
I tried to update my RTFQ Flip32 F3 OSD board from Cleanflight 1.12 to 1.14.2 on my RTFQ Flip32 F3 OSD board. Old Version in CLI said SPRACINGF3 so that is the firmware I installed. Flashing seemed to go OK, but now can't communicate with it. Red LED flashes 3 times fast, then twice slower. I even went back to the original version of CF. Same result. Then I tried Betaflight with the same result. I downloaded all the drivers, Betaflight shows Com3 is active. Then someone suggested I try other FC firmware. Flashing the SPRACINGF3EVO and the SPRACINGF3MINI will let me connect to the FC, but they don't work right. I set up everything in the configurator. But the ESCs beep once per second. The ESC instructions say this is no output from the throttle channel. And it seems the FC is not receiving anything from the receiver. In the configuration, I have the receiver in Configurator set to PWM, which is the type of receiver I am using. Do I need to set up one of the UARTS? Or is the FW not compatible with this FC? The SPRACINGF3 still does not allow me to connect at all. Configurator 1.8.7.

RipperDrone, 2017-01-15:
Sry, this is just an assumption, but the Omnibus.hex (=F3) firmware might be another target to try. It's based upon the Flip32 F3 as well. Not completely sure about gyro types and pinout details...

jflyper, 2017-01-15:
@aowi7280 
I think you are confusing Cleanflight and Betaflight. This is a Betaflight software issues page (and is NOT a support request page also).

Other things...
1. Betaflight configurator is not compatible with Cleanflight firmware (or vice versa).
2. The [RTFQ FLIP32 F3 OSD](http://www.readytoflyquads.com/flip-32-f3-osd) uses SPRACINGF3 target, flashing with other targets may result in hardware failures; don't try to flash it with anything other than SPRACINGF3 binaries.
3. Don't try to configure anything until you can connect to your board flashed with correct target binary using correct configurator. Doing so is a waste of time at least, and may toast your board, escs, motors, or you loose your fingers at worst.

RipperDrone, 2017-01-15:
@aowi7280 So your firmware flashed is still CF? In this case, flash BF first, then install latest BF configurator from Chrome store, then try to connect and test.

aowi7280, 2017-01-15:
Using CF configurator with SPRacingF3 target, the FC does not connect afterwards. So I tried BF configurator with SPRacingF3 target and the FC does not connect afterwards. Then BF with SPRACINGF3MINI or SPRACINGF3EVO target, it will connect, but the receiver does not function. I was thinking there was a bug in the SPRACINGF3 target with respect to this board.

RipperDrone, 2017-01-15:
Just to make sure we understnad: You re-flashed the FC with the SPRF3 target from BETAFLIGHT (v 3.0 or 3.1). Or was it still the CF flash firmware on the FC?

RipperDrone, 2017-01-15:
Flashing process itself was successfully completed?

RipperDrone, 2017-01-15:
If booting the FC (fast blinking, followed by 3 slow blinks and buzzer beeps) is ok, then it must be a mere receiver issue. You are using std PWM and no PPM/S-BUS protocol, correct?

In this case, you have to set the UARTS to Serial-RX = OFF and set the protocoll in BFC to PWM accordingly. Pls post your BFC screenshots of UART settings and receiver settings.

aowi7280, 2017-01-16:
Thanks for helping. Using betaflight 3.0.1, Config 1.8.8, and flashing with the FW loaded online was successful. I used the Load Firmware [online] button to get the SPRACINGF3.hex (SPRACING board/SPRACINGF3 3.0.1) After doing that, I have COM3 available in the top right corner of the BF configurator, but BF won't connect to the flight controller. Win10 Device Manager shows Silicon Labs CP210x UART to USB on COM3. I have uninstalled the driver, rebooted and reinstalled just to be sure it always comes up that way.

If I flash another target, like SPRACINGF3MINI, BF will connect to the flight controller, but receiver does not work. Yes PWM, no PPM/SBUS. I have not moved any wire connections since the copter was flying OK. All the Serial-RX is OFF in the Ports tab. In the configuration tab, the Receiver is set to PWM. Not sure if the screenshots will be helpful since this is not the correct target FW for this controller.

mikeller, 2017-01-16:
@aowi7280 I am wondering if what you've got there is really a 'FLIP32 F3 OSD' board, since these targets are sufficiently different that connecting to them shouldn't work if the wrong target has been flashed. Can you post a picture of your board?

RipperDrone, 2017-01-16:
Have u tried re-flashing with complete chip erase? Maybe there's something wrong with CPU load or settings of your SPRF3?

aowi7280, 2017-01-17:
[image]
Photo attached. STM32F303 processor, Sil CX2104, MAX7456 OSD
I did complete chip erase. The version I was running when the FC worked (before I tried to flash anything).
# version
# Cleanflight/SPRACINGF3 1.12.0 Jan 26 2016 / 06:55:21 (a83382e)

mikeller, 2017-01-17:
This is totally weird, since this board is clearly a non-VCP target (you can see the CX2104), but it can only connect if you flash a target that is VCP. I am not a windows driver expert, but to me this looks like it's a windows driver issue.

aowi7280, 2017-01-17:
Well, I installed all the drivers. I will try uninstalling all of them and starting over. I shouldn't need the STM driver or Zadig since there is a CX210x, right? I have a new OmnibusF4 on the way. Maybe it will work better.

jflyper, 2017-01-17:
@aowi7280 @mikeller @RipperDrone 
This board seems to require a special build.

Look around the postings in RCG
[LS-F3+OSD - SP Racing F3 Flight Controller With Integrated OSD](https://www.rcgroups.com/forums/showthread.php?2633525-LS-F3-OSD-SP-Racing-F3-Flight-Controller-With-Integrated-OSD/page26#post36478783)

blckmn, 2017-01-17:
Doesn't the SPRACINGF3 require a higher board rate when flashing?

https://github.com/betaflight/betaflight/wiki/Installing-Betaflight

> ** Only exception to this being for the SPracingF3 flight controller, for this FC you will need to check 'Manual baud rate' and change the selection to '230400' baud.

RipperDrone, 2017-01-17:
Have 5 SPRF3 running on BF, flashed them so many times over last year. Never needed to set the baud rate to 230k - strange...

jflyper, 2017-01-17:
@blckmn @RipperDrone 
The board is not SPRACINGF3. It's something else.

Ppl in above mentioned RCG thread all some local fork for CF or BF.

mikeller, 2017-01-17:
@aowi7280: I've added the FLIP32F3OSD target that should be right for your board to Betaflight proper. I don't have the hardware, so it'd be great if you could test it.

[betaflight_3.1.0_FLIP32F3OSD.zip](https://github.com/betaflight/betaflight/files/710283/betaflight_3.1.0_FLIP32F3OSD.zip)

aowi7280, 2017-01-17:
Awesome! It connects to Betaflight now. I will try to fly it as soon as I can.

jflyper, 2017-01-17:
@mikeller I took a quick look at the target.h, but couldn't find a substantial difference. What was it?

mikeller, 2017-01-17:
@jflyper I couldn't tell you what would have caused the connection problems - the wrong driver being used for the MPU causing an interrupt flood would be my guess. All I did was add the target definition in the fork linked in https://www.rcgroups.com/forums/showthread.php?2633525-LS-F3-OSD-SP-Racing-F3-Flight-Controller-With-Integrated-OSD/page26#post36478783 to 3.1.

Obviously SPRACINGF3 was never going to work for this target since it has the wrong MPU driver, so there is a justification to have it.

blckmn, 2017-01-17:
There's already a ZCOREF3 target that's essentially a SPRACINGF3 but with a MPU6500.

The drivers are written such that we could add the 6500 driver to the SPRACINGF3 target directly.

mikeller, 2017-01-17:
@blckmn:

Sorry, I wasn't aware of that. Also, I'd assume that targets that are variants of another target are implemented as variants, and not targets of their own, to reduce effort for maintenance.

Will look into renaming FLIP32F3 into ZCOREF3 and removing the stand-alone ZCOREF3.

mikeller, 2017-01-17:
Had a look, ZCOREF3 is yet different, since it uses SPI for the 6500 vs I2C on the FLIP32F3OSD. But still possible to convert it to a variant of SPRACINGF3.

Re adding multiple drivers, wouldn't that increase the target size, which is probably not what we want, since we're already pushing the limits of available space for F3?

aowi7280, 2017-01-18:
I got the latest configurator 1.8.9, flashed the FLIP32F3OSD target that mikeller attached above. But the motors won't go above an idle when using the radio. They work correctly using the Configurator motors control. Radio end points and subtrim are adjusted. All channels respond correctly in the Receiver tab. BTW, this board has a MP92 gyro IC.

# status
System Uptime: 520 seconds
Voltage: 115 * 0.1V (3S battery - OK)
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 1604
CPU:7%, cycle time: 507, GYRO rate: 1972, RX rate: 49, System rate: 9

# version
# Betaflight / FLIP32F3OSD 3.1.0 Jan 17 2017 / 20:20:14 (2be0d3f)
set min_throttle = 1120
set max_throttle = 2000
[image]

aowi7280, 2017-01-19:
Just tried the ZCOREF3 target and now the motors go to full speed. Something in the hex file above is not right.

mikeller, 2017-01-19:
There was a problem affecting a number of boards in this way a couple of days ago, and it looks like the custom build above is affected by it. It should be fixed by now, and I'll see if I can get the FLIP32F3OSD into the next RC, and otherwise do a custom build for you again, so you can do more testing.

aowi7280, 2017-01-23:
Thanks, I appreciate all the help.

closed 2018-02-23 by aowi7280

GotskGK, 2018-04-08:
Hi, I see that this issue has been closed and I would raise this issue on another website except that I am having the EXACT same issue as the person above described. I have an spf3 that was working absolutely fine, betaflight 3.2.5, until I had a fairly minor crash (only a few feet off the ground) and then when I plugged my battery in, the escs would keep on restarting and doing the beeping. I also received a series of beeps from the F.C. When I tried to connect to the F.C. on betaflight, it received Mo configuration within 10 seconds and closed the serial port. When I tried flashing, there was no response from bootloader. When I tried flashing in boot mode, it flashed successfully. When I tried connecting after that, I was still getting the 3 short beeps and two long beeps (the red light on the F.C. does the same as the beeper).

Any help would be greatly appreciated.

mikeller, 2018-04-08:
@GotskGK: Bad news, looks like your board got damaged (most likely the accelerometer / gyro unit) in the crash.