betaflight/betaflight #2126: CC3D Revolution, Arris ESCs & Motor response

NeoZeptepi, 2017-01-15:
Running Betaflight 3.1 on a CC3D Revolution flight controller.  Arris ESCs are running with PWM protocol, motor PWM speed separate from PID Speed; 500 frequency.  ESCs have been calibrated and all motors spool up correctly with variable throttle control from the Motors tab.  Min/max throttle values set.  TX channels are correct and test out.

When armed and the throttle brought up to the point that the motors engage, they spool up to different levels.  M1=50%, M2=100%, M3=idle, M4=50%.  Further throttle movement does not change motor speed; yaw stick movement has some effect, pitch/roll have no effect.

I'm posting here as the could be a bug.  Please let me know the best place to post if this is not appropriate.  Thanks.

jflyper, 2017-01-15:
@NeoZeptepi 
Did you do the motor test on the bench, without props?

If so, did you know this?
[Why do my motors keep accelerating on the bench when I arm without props ?](https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#why-do-my-motors-keep-accelerating-on-the-bench-when-i-arm-without-props-)

NeoZeptepi, 2017-01-15:
@jflyper Yes, it's been tested without props.  Motors work fine from BetaFlight.  Taranis works fine when watching channels from BetaFlight.  Could be some moving pictures are worth 1000 words; here is a quick video:

https://youtu.be/ZtLMpeEte7s

closed 2018-04-21 by DieHertz